raspberrypi/firmware #491: Crash when building PPSSPP for arm7 target - possibly RPI driver related?

joolswills, 2015-10-25:
Hi,

Was hoping to get your feedback on this as I wonder if it could be a driver issue on the RPI.

The emulator PPSSPP works on the RPI, but only if built for arm6. If -mcpu=arm7 is used (and with all arm7 specific asm and neon code disabled), it will crash. It crashes when compiling a shader it seems. I have no idea why and my knowledge of the code is very limited. Could it be related to the fact the RPI binary drivers are built for armv6 and the way data is aligned/passed etc?

joolswills, 2015-10-25:
woops. submitted too early - have updated the original issue above.

pelwell, 2015-10-26:
```
pure virtual method called
terminate called without an active exception

Program received signal SIGABRT, Aborted.
[Switching to Thread 0x73eff020 (LWP 2405)]
0x76a21f50 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56  ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt 
#0  0x76a21f50 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x76a23304 in __GI_abort () at abort.c:89
#2  0x76c25b5c in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#3  0x76c239a0 in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) thread apply all bt
```
Doesn't that spell out the problem? Calling a pure virtual method is illegal in C++, and the runtime has trapped that and generated an abort. The other threads look normal.

Calling a pure virtual function, assuming that the code has ever worked and isn't completely broken, is usually caused either by code only called in exceptional conditions (error handling, for example) or which isn't thread safe, with one thread attempting to use an object while it is being constructed or (more likely) destroyed. A Pi 2, with its 4 A7 cores, is going to stumble across race conditions that a Pi 1 might never hit.

joolswills, 2015-10-26:
As I'm not the author - no it doesn't spell out the problem :) But perhaps the authors can comment. 

The main thread is in the binary drivers at that point in time which looks related to me. I would add the software works fine on the rpi2 - just so long as it's compiled for armv6 and not arm7 (the problem is pi specific, and works on other multicore armv7 systems).

popcornmix, 2015-10-26:
This certainly isn't a firmware issue. Almost certainly a problem with the source code or way it's built.
Can you try a clean build? Changes to header files that don't trigger a recompile can certainly trigger this sort of error.

joolswills, 2015-10-26:
Yeh it was a completely clean build with debugging and -O0. I am stuck. If you think it's not possible that the GPU driver is causing the problem then I'm out of ideas. crashes at the same point every time. but recompile with -mcpu armv6 and all is fine. And there is no custom asmv7 asm or anything.

popcornmix, 2015-10-26:
No it's not going to be firmware related so I'll close this issue.
It's best to contact the author of PPSSPP. 
Seems to be a forum here: http://forums.ppsspp.org/

closed 2015-10-26 by popcornmix

joolswills, 2015-10-26:
There is a thread on the issue tracker already - I posted here as they thought it to be a Pi driver issue (see first post) and  https://github.com/hrydgard/ppsspp/pull/8107#issuecomment-150979391

popcornmix, 2015-10-26:
It's not crashing when compiling the shader. It's thread 6 that is crashing, and "pure virtual method called" is provoked by C++ code. None of the Pi specific libraries (in /opt/vc/lib) are written in C++.

joolswills, 2015-10-26:
it will be difficult to debug with such limited information in the backtrace for that thread.

pelwell, 2015-10-26:
What is the software environment being used here? "nptl/sysdeps/unix/sysv/linux/sem_wait.c" is a Chromium path.

That might also explain the truncated stacks...

joolswills, 2015-10-26:
I was using Raspbian Jessie - and gcc/g++ 4.9.x for that particular backtrace.

pelwell, 2015-10-26:
Ah - it's the Pthreads library.

My suggestion for debugging this would be to first create a really simple C++ program that calls a pure virtual function, and see how that appears inside gdb.

joolswills, 2015-10-26:
I would if it could be related to this in any way

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=42734#c29 (I see arm is referenced as having a problem here). Will try a few things.

joolswills, 2015-10-26:
(sorry - comment 40+ on that ticket references arm)

joolswills, 2015-10-26:
Will try your suggestion.

joolswills, 2015-10-27:
After reading through the GCC report above, and finding some other people with some similar issues,  it looks like it is some ABI compatibility issue with the armv6 Raspbian vs the armv7 compiled program. I switched to the Debian Jessie armv7 userland/libc6 etc, and it works fine - so there is a program with perhaps pthreads or libc6 built for armv6 on Raspbian linked with something built for armv7.

joolswills, 2015-10-27:
note I didn't recompile the program at all either. It just worked when Jessie was switched to armv7

unknownbrackets, 2015-10-27:
> It's not crashing when compiling the shader. It's thread 6 that is crashing, and "pure virtual method called" is provoked by C++ code. None of the Pi specific libraries (in /opt/vc/lib) are written in C++.

Well, unless there's memory corruption, or something...

We definitely do use atomics though (mentioned in that bug report as causing a pure virtual, if I understood correctly.)  We may be assuming that this platform has working atomics and trying to use them through gcc.

-[Unknown]

popcornmix, 2015-10-27:
Sounds like a gcc issue when mixing libs built for armv5 and armv7
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=42734#c49

joolswills, 2015-10-27:
or in this case, when built for armv6/armv7. I can confirm it now with libstdc++

```
pi@retropie ~ $ /opt/retropie/emulators/ppsspp/PPSSPPSDL 
43:47:737 Core/Config.cpp:898 I[LOAD]: Loading controller config: /home/pi/.config/ppsspp/PSP/SYSTEM/controls.ini
43:47:737 Core/Config.cpp:903 E[LOAD]: Failed to read /home/pi/.config/ppsspp/PSP/SYSTEM/controls.ini. Setting controller config to default.
Pixels: 1280 x 1024
Segmentation fault

pi@retropie ~ $ sudo dpkg -i ~/libstdc++6_4.7.2-5+rpi1_armhf.deb # built for armv7 from the gcc-4.7 sources on wheezy
dpkg: warning: downgrading libstdc++6:armhf from 4.8.2-21~rpi3rpi1 to 4.7.2-5+rpi1
(Reading database ... 62191 files and directories currently installed.)
Preparing to replace libstdc++6:armhf 4.8.2-21~rpi3rpi1 (using .../libstdc++6_4.7.2-5+rpi1_armhf.deb) ...
Unpacking replacement libstdc++6:armhf ...
Setting up libstdc++6:armhf (4.7.2-5+rpi1) ...

pi@retropie ~ $ /opt/retropie/emulators/ppsspp/PPSSPPSDL 
44:27:484 Core/Config.cpp:898 I[LOAD]: Loading controller config: /home/pi/.config/ppsspp/PSP/SYSTEM/controls.ini
44:27:485 Core/Config.cpp:903 E[LOAD]: Failed to read /home/pi/.config/ppsspp/PSP/SYSTEM/controls.ini. Setting controller config to default.
Pixels: 1280 x 1024
Virtual pixels: 1280 x 1024
I: gpu_features.cpp:124: GPU Vendor : Broadcom ; renderer: VideoCore IV HW version str: OpenGL ES 2.0 ; GLSL version str: OpenGL ES GLSL ES 1.00
glGetError 0x500

NO CRASH!
```

So it seems that this GCC bug really causes some problems for us with Raspbian targetted for armv6.

Time for a armv7 Raspbian perhaps ? Or perhaps there is a workaround for this problem in libstdc++ ? @popcornmix you are in contact with people who handle Raspbian - would be very appreciative if you could get some feedback.

popcornmix, 2015-10-27:
Realistically raspbian is not going to change anytime soon. It's not something we are in control of and supporting both Pi1 and Pi2 from the same sdcard is a very useful feature that won't be broken lightly.

So far of the millions of users, this is the only package I've seen which has this problem. I suspect finding out what is unique (e.g. in compile flags or libs used) that provokes the problem is the way forward.

pelwell, 2015-10-27:
In other words, just build for ARMv6.

joolswills, 2015-10-27:
don't think you can work around it with any compile flags.

According to the gcc issue - switching away from std::thread and using boost::thread with boost::atomic would solve it, but I doubt PPSSPP are going to do that though for the sake of Raspbian (There may be another fix for it of course).

I guess it will probably end up with the RetroPie project switching to Debian over Raspbian for the rpi2, which means more work for me.

I don't want to built for armv6 - as we lose all the NEON optimisations then.

pelwell, 2015-10-27:
Perhaps this is what you need: https://gcc.gnu.org/gcc-5/changes.html#libstdcxx

popcornmix, 2015-10-27:
If you say building for Debian Jessie armv7 userland/libc6 etc works, is it possible to build it statically, so distribution's libs are not used. Then the armv7 binary may run on raspbian (on a Pi2).

joolswills, 2015-10-28:
Thanks - will consider options. I have to automate everything, so I will come up with something that is the most reasonable to maintain.

joolswills, 2015-10-28:
Just to add, there are of course solutions available to me - but it would be useful if Raspbian had some armv7 targetted gcc - they could be in a 3rd party repository. Maybe that's the route I will take, as I realise I will have to sort this out myself. Thanks for the help though :)

joolswills, 2015-10-28:
Got a workaround!

```
pi@arrakis /usr/include $ g++ -dM -E - -march=armv7-a </dev/null |  grep "SYNC"
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_1 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_8 1

pi@arrakis /usr/include $ g++ -dM -E - -march=armv6 </dev/null |  grep "SYNC"
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
```

checking /usr/include/c++/4.7/ext/concurrence.h - requires __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 and __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 for atomic operations (which we are missing in libstdc++ with an armv6 target I am assuming)

so building with `-U__GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 -U__GCC_HAVE_SYNC_COMPARE_AND_SWAP_4` will force it behave like an armv6 and use _S_mutex

Thanks to the internet for pointing me in the right direction. (http://stackoverflow.com/questions/23583317/c-11-threads-error-pure-virtual-function-called)

unknownbrackets, 2015-10-28:
Nice find.

Strange that it wouldn't support a 4 byte compare and swap - can't that be implemented with LDREX/STREX (H, B, D) on armv6?  I guess it's just a library limitation?

-[Unknown]

joolswills, 2015-10-28:
I corrected my post above  - armv6 has __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 but not __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 it seems.