raspberrypi/firmware #492: Decoding corruption with h264

wm4, 2015-10-26:
Someone gave me this h264 sample: https://0x0.st/i2A.mkv

It starts playing fine. At one point, you see heavy corruption (lots of green) covering the whole screen. The corruption looks different with ffmpeg/mpv (which use MMAL) and omxplayer.

While mediainfo identifies the file as level 5, the transcoder producing this file is supposedly using x264 settings that put the stream within the RPI's hardware limits.

So I'm not sure whether this is really a bug. Maybe it actually does exceed what the hardware is capable of.

popcornmix, 2015-10-26:
Do you know when the corruption starts? It's 23 minutes long and no problems within the first few minutes...

wm4, 2015-10-26:
At timestamp 7:06 (0:38 if the player rebases start time) with MMAL, 6.57s with omxplayer.

Seeking back to the place of corruption does not always show it again.

popcornmix, 2015-10-26:
Just looks like a broken file to me. Both kodi and omxplayer seem to play the file for just over 5 minutes then exit. I've not seen any corrupt frames (although error concealment is enabled in those).

I tried with VLC on x86 Ubuntu and at 5:44 the file stopped (despite being reported as 23 minutes 37s long). I also can't seek past 5:44 with vlc. Are you sure the file is valid with any player?

wm4, 2015-10-26:
It's definitely broken with omxplayer on raspbian.

>I tried with VLC on x86 Ubuntu and at 5:44 the file stopped (despite being reported as 23 minutes 37s long). I also can't seek past 5:44 with vlc. Are you sure the file is valid with any player? 

The file reports 23 minutes duration, but is actually much shorter. Players showing this value merely trust the file header. Also, the start time is not 0, which makes things even more confusing.

To simplify this: corruption should show up within 1 minute of playback (if you don't seek). With libavcodec's h264 decoder no corruption occurs.

popcornmix, 2015-10-26:
I'm not seeing the corruption with kodi and mmal. But I have seen a mostly green screen with something changing in top-left corner with kodi and omxplayer.

I've also seen it with command line omxplayer. I don't think it occurred the first time I ran it, but I may have just been looking away.

I'll see if GPU shows anything with debugger attached.

popcornmix, 2015-10-26:
Okay, something unhappy in h264_cabac_run:
```
 587          // This asserts that our estimate for the size of the output was OK
 588          vcos_assert(actual_out_size <= wl->output_size);
```
We got 220672 out of cabac and estimated 154516 bytes.
Ping @deborah-c who will know more about this than me.

popcornmix, 2015-10-26:
Okay, this has been added to @deborah-c's queue to look at, but there are a few things ahead of that.

We did discover that adding 100K to cabac buffer size does fix this file, so I'm sure a fix is possible, just need to understand what is going wrong with the current size estimate.

julianscheel, 2015-11-19:
@popcornmix @deborah-c I fear I just ran into the same issue with this clip: http://jusst.de/files/corrupt-h264.ts
Could you have a quick look at whether it's the same issue? And if it is could you possible provide a testing firmware with increased cabac buffer size for now? As it is a major hit for us right now.
If it is a different issue I should probably move it into a dedicated issue.

popcornmix, 2015-11-19:
I've played file with debugger attached but it's not hitting the cabac size assert so I think this is a different issue - best create a new issue.
There is an assert at start:
```
4005    {/*coundn't find the reference picture, give up*/
4006       vd3_bs_assert(0);
4007       return;
4008    }
```
but I think that is expected as the file doesn't start from an I frame (VLC displays corruption for a while). Will need @deborah-c to investigate.

julianscheel, 2015-11-19:
@popcornmix I just opened a new issue: https://github.com/raspberrypi/firmware/issues/499
Regarding corruption on startup: The file is a live ts capture so for proper startup you'd have to rely on sei recovery being in place.

rcombs, 2016-01-09:
Any progress here? I hit this issue regularly.

rcombs, 2016-02-29:
_ping_; been a while with no updates here.

deborah-c, 2016-02-29:
Sorry, have been ill for most of the last year... I'm trying to get back to a state where I can work on things!

Ruffio, 2016-06-30:
@deborah-c how is it looking?

deborah-c, 2016-06-30:
I am dealing with a particularly unpleasant bout of ill health, I'm afraid