raspberrypi/firmware #493: mmal: Motion vectors on decode

julianscheel, 2015-10-30:
Would it be possible to provide the motion vectors of a h264 bitstream decoded via mmal to the userspace? Just like it can be done with the motion vectors generated by motion estimation on h264 encode via ```MMAL_PARAMETER_VIDEO_ENCODE_INLINE_VECTORS```?
I'm doing h264->mpeg2 transcode and it might be a great benefit to reuse the motion vectors from the h264 sources in the mpeg2 encoding process to lower the cpu load.

6by9, 2015-10-30:
Pretty certain not. 
On encode you have the CME doing the first pass motion estimation and that writes out to memory. It's the CME results that the INLINE_VECTORS mode spits out. 
I'm fairly certain that the decode path doesn't have such an option as there is not normally a need for it. If they were, then we could have passed those final MVs out from the encoder too as the decoder block has to be used to produce the reconstruction frame.
@deborah-c should be able to confirm that.

julianscheel, 2015-10-30:
@6by9 Thanks for explaining. Just another thought: Could there be a chance to make use of the CME alone without the full h264 encoding? So a software encoder could make use of it for the motion estimation step?

deborah-c, 2015-10-31:
Julian,

It is in theory possible; the decoder has to store motion vectors for reference frames for use in direct motion calculations, and at the cost of some extra memory one could do this for non-references as well. There isn't code to do this at the moment, though, and we'd need to find a way to pass the data back to the ARM (possibly as image metadata?)

Of course, the structure and quantisation of vectors differs between protocols, and figuring out which frame the vectors are relative to for each macroblock might be tedious, but it might still be useful.

This doesn't apply to encode, though, because the decoder in that use case doesn't decode actual bitstream, just internal representation (as far as I recall), and the symbol interpreter pipeline stage is the one that stores reference MV data. Since the encoder doesn't generate B-pictures, it doesn't have to worry about direct motion.

6by9, 2015-10-31:
And I learn something new :-)

The VC_IMAGE metadata doesn't get transferred to the ARM, but IL/MMAL buffers have a flags field, and currently the encode motion vectors uses the OMX_BUFFERFLAG_CODECSIDEINFO flag. That's not used in video_decode at the moment so could be used to signal MVs (though some sink components don't check for it, so video_render would need tweaking too to ignore it).
If the codec side is simple, then I can look at the IL/MMAL side. The codec could stuff it into VC_IMAGE metadata for IL to extract as a simple mechanism on the codec API.

julianscheel, 2015-10-31:
That sounds pretty good :)
I will dig around a bit with ffmpeg to see in how far the h264 motion vectors are usable for mpeg2 encoding, might need some tricky interpolation, but I can still think of it being useful.

@deborah-c Any thoughts on the possibility of using the motion estimation block alone? Maybe that would be even more beneficial for mpeg2 encoding than reusing the h264 vectors? Probably depending on how configurable that me block is.

6by9, 2015-10-31:
I guess others doing motion detection would like just the set of motion vectors without a video stream, so there are other potential uses.
Memory says that the CME was used in the past to get the motion vectors to be used for camera  image stabilisation, so there's no reason you can't have multiple clients. Controlling it isn't quite as tidy as one might hope - direct register writes from the codec code. 
The one bit I don't know is if/where interrupts of the CME are handled - normally there is a hardware interlock used between CME results and FME, so there's no need to enable the CME interrupts. I don't have the full datasheet to check that out.
NB The CME can only work on the luma if given planar data - it has to use a proprietary format to be able to work on the chroma as well. (In the H264 encoder it ignores the chroma regardless of format)

Purely from the point of view that it is a defined API, it would probably be easiest to wrap the CME as a MMAL/IL component (same as I have done for the CSI-2 receiver peripheral).
2 buffers in with the same timestamp, and one buffer out containing the MVs. It would have a strange output encoding though.
Data format would be the same as for the current motion vectors produced by the encoder EXCEPT not subsampled by a factor of two. How useful that is to you I have no idea.
Finding time for me to do it would be the issue :-(

julianscheel, 2015-11-01:
@6by9 Having an mmal component to do the motion estimation alone would be awesome. I've only had a short look into how motion estimation is done inside of ffmpegs mpeg2 encoder, but I think it should be possible to replace that with the data received from CME. I'll look into it in detail these days. Just in the hope that you find some spare time to implement the MMAL side sometime :)

F5OEO, 2015-11-01:
@julianscheel I planned also to try using motion estimation to help MPEG-2 encode. Have you a public repository for your project ? Maybe could contribute

julianscheel, 2015-11-01:
@F5OEO There is no code yet, but in would be ontop of ffmpeg git. First off I'm busy trying to understand how the motion estimation in ffmpeg mpeg2video encoder is done, so that it could be replaced my mmal generated motion vectors. You're welcome to look into that as well :)

F5OEO, 2015-11-01:
@julianscheel It seems to be implemented on omxcam project : https://github.com/gagle/raspberrypi-omxcam . I would like to look deeper how it works, but never have the time to. Think it could be a good approach !

Ruffio, 2016-08-10:
@julianscheel is this 'still an issue'? It looks like it is a suggestion/question and not a bug. Please consider closing this 'issue'.

julianscheel, 2016-08-18:
Can be closed for now. Didn't make it to a real project so far...

closed 2016-08-18 by julianscheel