raspberrypi/firmware #494: Firmware mangles device tree clock-frequency variable type

ali1234, 2015-11-05:
I conducted a small experiment to see how the firmware modifies the flattened device tree blob before handing it to the kernel. I used u-boot, tagged with mkknlimg and dumped the device tree blob loaded at 0x100 by the firmware. Then I loaded the same firmware from the fat partition with u-boot and dumped it again. The I diffed the two dumps. The relevant part of the diff:

```
< 			clock-frequency = <0x2faf0800>;
---
> 			clock-frequency = "5��";
448c479
< 			clock-frequency = <0x2faf0800>;
---
> 			clock-frequency = "5��";
456c487
< 			clock-frequency = <0x2faf0800>;
---
> 			clock-frequency = "5��";
464c495
< 			clock-frequency = <0x2faf0800>;
---
> 			clock-frequency = "5��";
```

The clock-frequency variables were 0x2faf0800 (=800000000) in the original blob. The firmware changed them to "5��", which is 0x35a4e900 (=900000000) interpreted as a string. So I think that the firmware not only changed the value but also changed the type of this variable to a string, and based on the new value I believe this was unintentional.

pelwell, 2015-11-05:
The underlying FDT representation is untyped, or rather all properties are just arrays of bytes open to interpretation. The dump utilities (fdtdump, dtc -I dtb, etc.) have to guess at the types of property based the byte values they contain. In this case I would imagine that it sees the unprintable '\x08' in 800,000,000 and assumes it must be an integer, whereas '\x35\xa4\ze9\x00' could be an 8-bit ASCII string.

ali1234, 2015-11-05:
Spot on. It actually tests the string with isprint().

closed 2015-11-05 by ali1234