raspberrypi/firmware #495: cec device know longer detected with LCD screen

pssc, 2015-11-06:
root@screenpi:/home/pi# /opt/squeezeplay/bin/libcec-daemon -l -v
INFO - Opened /dev/uinput
INFO - Created uinput device
ERROR - No adapters found
root@screenpi:/home/pi# uname -a
Linux screenpi 4.1.12-v7+ #824 SMP PREEMPT Wed Oct 28 16:46:35 GMT 2015 armv7l GNU/Linux
root@screenpi:/home/pi# 

This worked when I first got the screen going with simultaneous HDMI output (audio still working)

popcornmix, 2015-11-06:
Do are you saying this worked with an earlier firmware version? If so can you identify the version that stopped it working?
(using rpi-update <hash> with the hash from https://github.com/Hexxeh/rpi-firmware/commits/master)

pssc, 2015-11-07:
Couldn't  find one that worked chopped through to September, however I can not confirm that even with display_default_lcd=0, I cant get cec working, confirmed it still works with the B on my desk.

popcornmix, 2015-11-07:
Does unplugging the LCD make CEC work?

pssc, 2015-11-07:
No... which probably means libcec library issues...

popcornmix, 2015-11-07:
You say the B works? I assume that's with a different TV? Does swapping the Pi's over change anything? Has the non-working TV ever worked? Might be worth a full power cycle of TV (unplug from mains for a while) - they often seem to retain stale information about a CEC client which stops it working.

pssc, 2015-11-07:
no same TV swapped the HDMI cable to the B

popcornmix, 2015-11-07:
Are you running raspbian? Same versions or is Pi2 newer?
Can you swap sdcards between the two Pi's?

pssc, 2015-11-09:
@popcornmix confirmed as a libcec2 issue not all versions of libcec2 are the same libcec2 >= 2.2.0 was needed on jessie(2.1.4) to make the tool report the inbuilt adaptor static linking to libcec2 wasn’t enough either, sorry about the wild goose chase.

popcornmix, 2015-11-09:
So is this sorted? Okay to close?

closed 2015-11-09 by pssc