raspberrypi/firmware #500: RPI2 freezes when music plays using mpd

hyunchulnah, 2015-11-20:
Hi everyone.

I'm using mpd on a headless RPI2 with Cirrus Logic sound card (Jessie kernel 4.1.10-v7+),
and I have been having a headache because it completely freezes quite often.

The symptoms when freezing are
  1. sound also freezes : I can hear sound though it is the endless repetition of short (<1sec) duration
  2. no response on ssh, ping from outside : only way to stop is unpluggin power
  3. no meaningful log remained : both on mpd.log and system log

It is very similar to [this](https://github.com/raspberrypi/firmware/issues/247) or [this](https://github.com/raspberrypi/firmware/issues/263) except that in previous cases, the music plays according to playlist, while endless repetition of short sound in my case.

It seems to be attributed to the USB firmware issues in the previous cases, however, I do not use USB storage.
I use network storage server as NFS mount. I had also tried SMB mount in vain.

Interestingly, I can control using Ampache on the storage server (Localplay) and no freezing has been occurred yet.
Since Ampache streams music to mpd, I suspect the problem comes from around the network storage mount without further progress.

Is there any idea or clue?

Any comment will be appreciated.

Thank you for your reading.

popcornmix, 2015-11-21:
Are you overclocking? If so disable that and report back.

hyunchulnah, 2015-11-22:
I do not use overclocking. At raspi-config, None for Overclocking.

Thank you.

popcornmix, 2015-11-22:
Are you using this I2S soundcard: http://www.element14.com/community/community/raspberry-pi/raspberry-pi-accessories/cirrus_logic_audio_card

Can you switch to the internal analog audio (headphone socket on the pi) and confirm if the hang still occurs? That would determine if it's related to the cirrus driver or now.

hyunchulnah, 2015-11-24:
Sorry for late reply.

Yes. I use that soundcard and driver was set-up following https://stmllr.net/blog/cirrus-logic-audio-card-on-raspberry-pi2-with-debian-jessie/index.html.

By the way, could you please tell me or give some related links how can I switch sound output to internal audio instead of the sound card?

Thank you.

hyunchulnah, 2015-12-01:
Originally I set-up cirrus drivers on Raspbian jessie release directly. But the author of blog https://stmllr.net/blog/cirrus-logic-audio-card-on-raspberry-pi2-with-debian-jessie/index.html recommends (for the time being) to start from wheezy release for some reasons.

So I experimented new driver set-up from wheezy to jessie dist-upgrade procedure last week.
No freezing so far. I need more time to check whether it really works or not.

hyunchulnah, 2015-12-07:
Still no freezing. Seems to be fixed.
Thank you.

popcornmix, 2015-12-07:
Okay to close?

hyunchulnah, 2015-12-08:
Yes. Please.

closed 2015-12-08 by hyunchulnah

tkalinows, 2016-02-21:
Hi.
I'm glady. It's seems that problem is in sendmail. reinstall sendmail give succes. RPI not freeze for 5 days.
Check in your case.
best regard

hyunchulnah, 2016-02-22:
Hi tkalinows.

In fact, I did not use sendmail. Instead, Exim4 was installed as a default MTA.

Do you mean exim4 by sendmail?

Thank you.

tkalinows, 2016-02-23:
Just try apt-get remove sendmail and all part.
Maby in your case it do not help, but You can try :)
best regards!