raspberrypi/firmware #501: Unable to use LCD as output for omxplayer when HDMI is primary display

reufer, 2015-11-20:
According to the description of [dual display usage](https://www.raspberrypi.org/blog/the-eagerly-awaited-raspberry-pi-display/), it should be possible to make HDMI default by adding "display_default_lcd=0" to config.txt and the call omxplayer with "--display=4" to show videos on the LCD.

However this does not work and omxplayer somehow gets stucked. The behaviour is the same with my application, so I assume it's a firmware issue.

popcornmix, 2015-11-20:
Does booting on LCD and then running omxplayer with --display=5 work for you?

reufer, 2015-11-20:
Yes, the "normal" case works as expected - I just wanted to test it the other way around, however it might be a bit unusual.

popcornmix, 2015-11-23:
Just tried and it works fine. What does:
```
vcgencmd get_version
omxplayer --version
```
report?

reufer, 2015-11-24:
Just did a apt-get update/upgrade and reboot, but the problem still exists:

```
pi@raspberrypi ~ $ omxplayer -o local --display=4 test.mp4 
Video codec omx-h264 width 720 height 576 profile 100 fps 50.000000
Audio codec ac3 channels 6 samplerate 48000 bitspersample 16
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 720x576@50.00 interlace:3 deinterlace:1 anaglyph:0 par:1.45 layer:0 alpha:255
```
That's it, then omxplayer is freezed, but other shells are not affected. Omitting `--display=4` plays the clip on HDMI just fine.

```
pi@raspberrypi ~ $ vcgencmd get_version
error=1 error_msg="Command not registered"
pi@raspberrypi ~ $ omxplayer --version
omxplayer - Commandline multimedia player for the Raspberry Pi
        Build date: Sat, 12 Sep 2015 00:49:19 +0000
        Version   : d99bd86 [master]
        Repository: https://github.com/popcornmix/omxplayer.git
pi@raspberrypi ~ $ vcgencmd get_mem gpu
gpu=256M
```

Power seems to be ok, there's no colored square visible nor it's disabled. My config.txt looks like this:
```
gpu_mem=256
lcd_rotate=2
display_default_lcd=0
decode_MPG2=0xae2aef52
disable_splash=1
```

popcornmix, 2015-11-24:
Sorry - should have been ``vcgencmd version``

reufer, 2015-11-24:
```
pi@raspberrypi ~ $ vcgencmd version
Nov 18 2015 15:43:45 
Copyright (c) 2012 Broadcom
version 0c9af7cde38e223c95c3404f468db8fc5004495c (clean) (release)
```
Disabling ```lcd_rotate=2``` does not make any difference.

popcornmix, 2015-11-24:
Okay, I've reproduced. It wasn't happening with a debug build of firmware, but I can see with a release build and have reproduced with a modified debug build.

popcornmix, 2015-11-24:
Does this firmware help:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_lcd.zip

reufer, 2015-11-24:
Looks good to me! :-)

popcornmix, 2015-11-25:
This fix is in latest rpi-update firmware. Please test and close if all okay.

reufer, 2015-11-26:
Perfect, thanks a lot!

closed 2015-11-26 by reufer