raspberrypi/firmware #502: Feature Request: HDMI hotplug

imzyxwvu, 2015-11-22:
Add an option in config.txt to power on HDMI automatically when HDMI is plugged in. This would add great convenience if someone needs to hotplug HDMI. After HDMI is plugged in, we need an SSH access to execute `tvservice -p`. For most modern computers, when HDMI is plugged in, it would be automatically configured and powered on.

popcornmix, 2015-11-22:
I suspect this wouldn't do any good apart from stopping composite output.

``tvservice -p`` isn't sufficient to enable HDMI output usefully. The console framebuffer, or X or an application using dispmanx layers (e.g. omxplayer or kodi) won't be able to display anything without creating a new framebuffer / dispmanx elements of the right size. That generally means restarting any apps that are using the display, and reconfiguring the framebuffer (e.g. with fbset).

If you boot without hdmi attached the framebuffer will be 656x416 and output to composite. Outputting that framebuffer to a 1920x1080 HDMI display wouldn't be satisfactory and you would have to restart X or whatever application you are running to make use of the new resolution.

Is your issue that you want to switch dynamically between composite and hdmi displays?
If not, then ``hdmi_force_hotplug=1`` in config.txt is probably the best option.

imzyxwvu, 2015-11-22:
Yes, what I need is to turn on HDMI automatically when the system is already up. My use case is that I need to share one monitor between my Pi and my PC by hotplugging the HDMI cable. When HDMI is plugged into my PC, the monitor immediately turned on and the desktop appears.

I didn't want to switch to composite output since the sizes of their framebuffers are not same, so just leave composite off when this option is set.

popcornmix, 2015-11-23:
So if you add ``hdmi_force_hotplug=1`` what is the problem?
I would imagine if you unplugged the monitor from PC and plugged it into the Pi you would see the Pi''s HDMI output.

imzyxwvu, 2015-11-23:
Thanks, that worked. Sorry for bothering.

closed 2015-11-23 by imzyxwvu