raspberrypi/firmware #503: QPU deinterlace priority vs GL application priority

julianscheel, 2015-11-25:
Is it possible to influence the priority of the QPU advanced deinterlace for HD content compared to other GL applications?
I do render an OSD with Qt, using EGL, which generally works fine. Now when I enable the new HQ HD deinterlacing on the QPU (Param 3 is set to 0 for full frame rate, Param 4 is set to 1 to enable QPU usage) I get lots of frame drops while my OSD fades in or out. Sometimes it even leads to the decoder being stuck and then running in ENOSPC errors.
Now I wonder if it would be possible to lower the priority of the OSDs GL code so that the video would not be affected? If the OSD runs with reduced framerate at that point it wouldn't hurt much.

popcornmix, 2015-11-25:
1080i60 advanced deinterlace is ridiculously expensive and takes the vast majority of QPU and sdram bandwidth, so options are limited.
Kodi by default limits the framerate of GL updates to 10fps when playing video which seems to work okay. Are you able to limit the submitted GL frames and is that sufficient?
Another option is to enable half rate deinterlace when the gui is busy and switch back to full rate when gui is removed.

julianscheel, 2015-11-25:
Limiting the OSD fps seems like a reasonable option. It seems Qt5 does support this, I'll give it a try tomorrow, thanks.

julianscheel, 2015-11-25:
Do you actually require overclocking for 1080i QPU deinterlace to work with omxplayer on the Pi 1, btw?

popcornmix, 2015-11-25:
In theory advanced deinterlace of 1080i60 just fits into 250MHz qpu budget (just under 16ms per frame).
However due to other users of sdram you do get occasional slower frames which can cause frame drops. For this reason there is an automatic boost to 300MHz of 3D clock which gives a bit of margin. It still can get tight if there is a lot of sdram of other 3D work going on.

julianscheel, 2015-11-25:
Ok. Which clock is qpu driven by? Is it v3d? So manually overclocking to 350 or 400MHz would give some additional headroom in case the Pi can handle that stable?

popcornmix, 2015-11-25:
v3d. Run ``vcgencmd measure_clock v3d`` to confirm.
Yes, setting v3d_freq=400 would help. 500 is stable for me on most Pi's.
(personally I'd just set gpu_freq to set all clocks as there are fewer plls than clocks and it avoids the risk of accidentally underclocking another gpu clock).

julianscheel, 2015-11-25:
Surprisingly v3d is reported as 0 by measure_clock. I'll try to set gpu_freq to 400 or 450 then tomorrow and see how it goes (along with limiting framerate of the OSD).

popcornmix, 2015-11-25:
measure the clock when busy to get a correct reading. It may well read 0 when idle.

julianscheel, 2015-11-25:
Ah right, I see. Assumed an open GL app would be sufficient. It's running at 250 MHz right now, but QPU deinterlace is disabled. So let me play around a bit tomorrow!

popcornmix, 2015-11-25:
If the GL app is just doing GUI stuff you may have to catch it at the right time. It may well only be 10% utilised. v3d does get enabled/disabled each frame.

julianscheel, 2015-11-26:
If I set gpu_freq to 450 the v3d clock still does not raise above 333 MHz while the deinterlacing is running:
```
# vcgencmd get_config int
arm_freq=950
config_hdmi_boost=2
core_freq=500
disable_auto_turbo=1
disable_overscan=1
disable_splash=1
emmc_pll_core=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=450
hdmi_force_cec_address=65535
over_voltage=6
pause_burst_frames=1
program_serial_random=1
ramfsaddr=0x900000
sdram_freq=400
temp_limit=85
# vcgencmd measure_clock v3d
frequency(43)=333333000
```
Any thoughts why that would be? Possibly some unresolvable relation due to how I set the other frequencies? Is it actually possible to see the whole clocking tree, btw?

popcornmix, 2015-11-26:
Remove core_freq=500 or update the firmware.
Older firmware Used same PLL for core, v3d, h264, isp and that means they are all dependend.
Recent firmware uses a dedicated PLL for core and a shared one for v3d, h264, isp.

What's happening is core is the master. 500MHz produces a 2GHz PLL. Other clocks need an even divisor of PLL. So 2GHz/6 = 333MHz.

julianscheel, 2015-11-26:
Ok, that works, thanks. What I find kind of suprising is that the CPU load raises by ~25% if I enable QPU deinterlace for 1080i50 sources. With old deinterlacing it is roughly 50% cpu load playing an mpeg ts with vlc when running with arm_freq 950MHz. If I enable QPU deinterlace it raises to 75%. Is there actually any extra load for the deinterlacing on the CPU? Or would you assume this to be solely busy waiting due to the longer it takes to get deinterlaced frames out of the system?

popcornmix, 2015-11-26:
The slowdown will be due to sdram latency increasing when busy. No extra code runs on the arm.

julianscheel, 2015-11-26:
Ok, I might try to overclock sd a bit more. But I just recently reduced it from 500 to 400 as 500 was unstable with many of the Pis with Elpida ram...

popcornmix, 2015-11-26:
Yes, sdram_freq of 500 or even 600 is often fine on a pi1.
But Pi2 has external sdram which limits its sdram overclock. 500 is often unstable on pi2.

julianscheel, 2015-11-26:
Unfortunately even the latest Pi1 production we've received is not stable with 500 anymore. They have the Elpida memory on them. The Samsung ones are very robust indeed.

popcornmix, 2015-11-30:
Out of interest where did you get the Elpida Pi1 from? What model of Pi was it?

Would you be interested in swapping it for an alternative Pi1 (with Samsung memory) or a Pi2?
I wouldn't mind running a few tests on it to see if any settings can be tweaked to make 500MHz reliable.

julianscheel, 2015-12-01:
It appears my memories got a little bit shuffled. It's not actually Elpida but Micron memory. I think Elpida got acquired by Micron?

I can't tell for sure but those should have been delivered by Farnell and it's a Model B. We checked our stock and all the rest (same delivery) appears to be Samsung, so only a few Microns have been inbetween of which all but one were delivered to customers already. But if you think it's worth playing around with it I can ship that one straight to you.

popcornmix, 2015-12-01:
If it fails easily with sdram_freq=500 then I'm still interested.
email me at popcornmix at gmail dot com with details of where to send a replacement.

Ruffio, 2016-06-29:
@julianscheel can this issue be closed?

julianscheel, 2016-06-29:
@Ruffio yes, sorry, forgot about this ticket.

closed 2016-06-29 by julianscheel