raspberrypi/firmware #504: RTC doesn't work. i2c_bcm2708 doesn't get loaded

DougieLawson, 2015-11-25:
I've got
```
dtparam=i2c=on,spi=on
dtoverlay=i2c-rtc,ds1307
#dtparam=spi=on
#dtparam=i2c_arm=on
dtoverlay=sdhost
gpu_mem=16
initramfs initrd.sda
max_usb_current=1
```
in /boot/config.txt

Running 
Linux apollo 4.1.13+ #826 PREEMPT Fri Nov 13 20:13:22 GMT 2015 armv6l GNU/Linux
with a I2C DS1307 real time clock.

At boot time i2c-dev and rtc-ds1307 both get loaded but i2c_bcm2708 doesn't. That means the RTC isn't active until I modprobe that missing module

```
root@apollo /var/log # cat messages | grep rtc
root@apollo /var/log # i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
root@apollo /var/log # modprobe i2c_bcm2708
root@apollo /var/log # i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

The i2c driver needs to be automatically loaded when an I2C RTC device is defined in config.txt.

pelwell, 2015-11-25:
Did you forget to remove `blacklist i2c_bcm2708` from /etc/modprobe.d/raspi-blacklist.conf (and all other .conf files in that directory)?

DougieLawson, 2015-11-25:
Arrgh, who left that junk on my system.

closed 2015-11-25 by DougieLawson