raspberrypi/firmware #505: adding return_address to config.txt

roman3017, 2015-11-27:
In order to explore TrustZone with OP-TEE OS on RPI, it would be nice if there was an option to specify a return_address in config.txt, which would be loaded to LR register before jumping (bx) to the kernel.

That way we could load optee and a non secure code to the memory using existing attributes from config.txt. Then we could run optee first, which would at the end switch to the non secure mode and jump to the address stored in the LR register.

popcornmix, 2015-11-28:
I'd suggest you run with kernel_old=1 and prepend whatever startup code you desire to the kernel.
Execution will start from address zero of your kernel.img file.
I'd suggest:
0x0 - 0xff boot code
0x100->0x8000 leave free for kernel ATAGS
0x8000-> your kernel code.

Then you are free to set LR (and any other regs) to whatever you require before branching to kernel start. See https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=100752 for more details on boot code.

roman3017, 2015-11-30:
Ok, thanks. You are suggesting to modify OP-TEE, which would run before the kernel instead. I was hoping that you could enable unmodified OP-TEE by adding the attribute, since it expects the kernel address in LR register.

popcornmix, 2015-11-30:
You shouldn't need to modify the OP-TEE image.
You should assemble the entry.S code from [here](https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=100752). Create a padding file of zeros of length 0x8000-(length of assembled entry.S).
Concatenate entry.bin+padding.bin+optee.bin into kernel.img.

roman3017, 2015-11-30:
I see, thank you very much. This sounds workable.

closed 2015-11-30 by roman3017