raspberrypi/firmware #506: !HDMI => Composite enabled by default. Relevant for Pi Zero?

P33M, 2015-11-28:
As I currently understand it, by default the Composite TVDAC will be enabled when

- There is no HPD signal from the HDMI port
- No HDMI settings are prescribed in config.txt
- EDID doesn't respond despite HPD not being asserted

Is this sensible behaviour for Pi Zero, given that the Composite output is a pair of through-hole pads with no connector attached?

We can shut down a PV and PLLH by default if we assume that Composite is not connected.

popcornmix, 2015-11-29:
One issue is if there is no EDID we don't know what resolution to drive.
VGA60 is the only mandatory HDMI resolution, but (certainly if booting to desktop) that is not really usable. We could try 720p60 but it may not be supported (especially by some computer monitors).
I'm not sure I agree with special casing the Pi Zero just because the composite output is less accessible (but still supported).
A different question is what percentage of user are using composite? I'd guess around 5%, but it may be lower and if so we could consider making HDMI the default and composite users must edit config.txt.

JamesH65, 2017-12-21:
@P33M @popcornmix Been a long time, ready to close?