raspberrypi/firmware #507: Video decoder detects wrong pixel aspect ratio

reufer, 2015-11-30:
I have some SD-MPEG2 streams with 16.9 aspect ratio. VLC plays them fine and ffmpeg reports correct dimensions:
```
Stream #0:0(eng): Video: mpeg2video (Main), yuv420p(tv), 720x576 [SAR 64:45 DAR 16:9], max. 8500 kb/s, 25 fps, 25 tbr, 1k tbn, 50 tbc (default)
```
However, omxplayer detects a curious pixel aspect ratio of 64/27 and thus shrinks the video vertically:
```
V:PortSettingsChanged: 720x576@25.00 interlace:3 deinterlace:1 anaglyph:0 par:2.37 layer:0 alpha:255
```
Do you have an explanation for this? I've uploaded a sample here: www.reufer.ch/temp/aspect.mkv

popcornmix, 2015-12-03:
I think this is an omxplayer issue, rather than firmware.
Kodi plays the file correctly (in dvdplayer or omxplayer mode).

Mediainfo says:
```
Display aspect ratio                     : 16:9
Original display aspect ratio            : 2.40:1
```
Not quite sure what it means by that, but I suspect that the two DARs is the reason for the discrepancy. I'll see if I can trace what information kodi is using, compared to the information omxplayer is using.

popcornmix, 2015-12-03:
Can you try omxplayer with new commit: https://github.com/popcornmix/omxplayer/commit/89c4b1639df8bf1a9c6d9bf8dfde0252fec69163

reufer, 2015-12-04:
I'm not sure if this is the right way. First of all, my application uses plain TS, so there's no matroska header to check. Second, the sample I provided was originally plain TS, converted to mkv by ffmpeg for a quick test with omxplayer - so the correct parameter must have been inside TS already. In my view, the MPEG2 decoder just reports wrong aspect ratio.

popcornmix, 2015-12-04:
Can you provide the ts file?

reufer, 2015-12-04:
Sure: www.reufer.ch/temp/aspect.zip
You can either extract the archive in VDR's video directory as descried in https://github.com/raspberrypi/firmware/issues/463#issuecomment-158791734 or look for 00001.ts inside the folder %Stromberg/2004-10-11.20.15.1-1.rec.

popcornmix, 2015-12-04:
This will probably need our codec expert (@deborah-c) to comment.

Parsing seq->seq.aspect_ratio_information gives 3 which is a DAR of 16:9.
The code then does:
```
         // The values we have are DAR, we need to convert to SAR
         if (seq->have_seq_display_ext)
         {
            aspect_y *= seq->seq_display_ext.display_horizontal_size;
            aspect_x *= seq->seq_display_ext.display_vertical_size;
         }
         else
         {
            aspect_y *= seq->vd3_seq.crop_width;
            aspect_x *= seq->vd3_seq.crop_height;
         }
```
Using seq->vd3_seq.crop_width/seq->vd3_seq.crop_height does give the desired result (seq->vd3_seq.crop_width=720 seq->vd3_seq.crop_height=576).

But seq->seq_display_ext.display_horizontal_size = 540 and seq>seq_display_ext.display_vertical_size = 576. The 540 value is suspicious from the stream. Possibly an encoder error, or something I don't understand.

However it seems wrong to be using that here. We know the shape we want to the frame to be (16:9). We know how many pixels we will send to host (720x576). Host is going to scale the 720x576 pixels by PAR hoping to get 16:9 out.

Using  display_horizontal_size (which the host doesn't know about) to determine PAR seems wrong to me.

reufer, 2015-12-04:
Thanks for the detailed analysis. I just checked ffmpeg's parser and display extension is not used at all. Possibly ffmpeg is just too lazy to be fooled by an erroneous stream?

popcornmix, 2015-12-05:
If you want to test my proposed firmware change, try:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_aspect.zip
No guarantee it won't cause regressions, but I'd be interested if it fixes your current problem, and if you can spot any problems.

reufer, 2015-12-06:
Looks good - thanks a lot! I haven't seen any side effects so far, but I'll keep testing.

popcornmix, 2015-12-08:
I have added this to official firmware and will keep an eye out for any bug reports...

popcornmix, 2016-01-17:
Okay to close this issue?

reufer, 2016-01-17:
Sure, I haven't seen any side effects. Thanks a lot!

reufer, 2016-03-18:
I'm afraid I found a stream where your solution does not work:
http://www.reufer.ch/temp/heitere.ts

The decoder reports pixel aspect ratio of 16/15, however it's a 16:9 video. I'd appreciate if you could have a look at it. Thanks a lot!

JamesH65, 2017-12-21:
@reufer Is this still an issue? Was it a faulty stream?

reufer, 2017-12-21:
I haven't tested recently, but you'll find the stream [some lines above](https://github.com/raspberrypi/firmware/issues/507#issuecomment-198494810). I don't think the stream is faulty, as far as I remember, ffmpeg reported correct dimensions.

deborah-c, 2017-12-21:
I can look at this (and the other decoder-related issues) after 
Christmas, assuming Eben would still like me to. Please drop me an 
email!

Deborah

<mention@noreply.github.com>
Sent: 21/12/2017 15:37:01
aspect ratio (#507)