raspberrypi/firmware #508: Please add lm75 kernel module

iz8mbw, 2015-11-30:
Hi.
Please is it possible to add the lm75 (lm75.ko) kernel module?
In hwmon it's not present.

root@raspberrypi:/lib/modules/4.1.13+/kernel/drivers/hwmon# ls -la
total 40
drwxr-xr-x  2 root root  4096 Nov 27 11:21 .
drwxr-xr-x 41 root root  4096 Nov 27 11:22 ..
-rw-r--r--  1 root root 10871 Nov 27 11:21 hwmon.ko
-rw-r--r--  1 root root  7370 Nov 27 11:21 sht21.ko
-rw-r--r--  1 root root  8588 Nov 27 11:21 shtc1.ko
root@raspberrypi:/lib/modules/4.1.13+/kernel/drivers/hwmon#

Thanks!

Ruffio, 2016-08-10:
@popcornmix can this be achieved?

iz8mbw, 2016-08-11:
For what I know lm75 is still not present

6by9, 2016-08-11:
The request should really be in https://github.com/raspberrypi/linux/issues as it is a kernel module, not a firmware change.

CONFIG_SENSORS_LM75=m
The module comes out at 12875 bytes on a 4.4.15 tree.
Does it need a device tree overlay too, although there doesn't appear to be one for the sht21 or shtc1 modules currently built. Perhaps hwmon sorts this out.

popcornmix, 2016-08-12:
This is now in latest ``rpi-update`` kernel.

iz8mbw, 2016-08-13:
Thanks.
LM75 driver works fine.
Hope it will be still available also in future rpi-update.
Issue can be closed.
Again thanks!!

closed 2016-08-13 by popcornmix