raspberrypi/firmware #512: Low Resolution Snapshot Error

ixs0792, 2015-12-03:
There seems to be an error in the vc_dispmanx_resource_create, vc_dispmanx_rect_set, vc_dispmanx_snapshot, vc_dispmanx_resource_read_data command chain that causes some videos to produce incorrect results when using low width downscaling.

I am able to recreate the error using [raspi2png] (https://github.com/AndrewFromMelbourne/raspi2png) (and others) to output 32x18 files. 
When playing a video via omxplayer that is "Video: h264 (Main)" encoded it works fine.
However, playing a video that is "Video: h264 (High)" encoded it produces a nearly black output.

Switching to 64x36 output both videos work fine.

This doesn't seem too critical (a 2:1 downscale is easy enough), but it drove me crazy for a few days trying to figure it out...

popcornmix, 2015-12-03:
Pretty sure the profile/level of the video doesn't matter. The resolution does. What is the resolution of the problematic video file?
What is the command (using raspi2png) you enter that shows the problem?

ixs0792, 2015-12-04:
You might be right. I forgot that I also changed the video size when I re-encoded it. 

The original file omxplayer -i came back with:
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'source/altj.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: isommp42
    creation_time   : 2015-04-25 09:55:19
  Duration: 00:02:50.34, start: 0.000000, bitrate: 2340 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1280x720 [SAR 1:1 DAR 16:9], 2146 kb/s, 23.98 fps, 23.98 tbr, 24k tbn, 47.95 tbc (default)
    Metadata:
      creation_time   : 1970-01-01 00:00:00
      handler_name    : VideoHandler
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 191 kb/s (default)
    Metadata:
      creation_time   : 2015-04-25 09:55:20
      handler_name    : IsoMedia File Produced by Google, 5-11-2011
have a nice day ;)

After I recoded it via Handbrake the info is:
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'source/left.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    creation_time   : 2015-11-29 17:07:20
    encoder         : HandBrake 0.10.2 2015061100
  Duration: 00:02:50.37, start: 0.000000, bitrate: 605 kb/s
    Stream #0:0(und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709), 320x180, 435 kb/s, 23.98 fps, 23.98 tbr, 90k tbn, 180k tbc (default)
    Metadata:
      creation_time   : 2015-11-29 17:07:20
      handler_name    : VideoHandler
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 163 kb/s (default)
    Metadata:
      creation_time   : 2015-11-29 17:07:20
      handler_name    : Stereo
have a nice day ;)

The ./raspi2png -w 32 command on the 1280 wide version returns a black frame, but ./raspi2png -w 64 return what I would expect. I checked the other videos I've been testing with and all the video <1280px wide work fine.

popcornmix, 2015-12-05:
There seems to be a factor of 32 decimation limit. 1280->40 (factor 32) is okay. 1280->38 (just over 32) appears very dark. Not sure if this is a HVS hardware limit.

As you say, this is only an issue for very small destination images, so decimating by 2 in software shouldn't be expensive.

I'll try and look into this more at some stage, but it's pretty low priority.

ixs0792, 2015-12-05:
The hard part was figuring out that the issues wasn't in the way I was trying to use vc_dispmanx... Once I figured that out writing the 2:1 decimator to work around it was trivial. 

I agree this is very low priority, but I figured I would let people know it exists.