raspberrypi/firmware #513: firmware updating older installs for pi zero ->

ladyada, 2015-12-10:
hiya, not sure if this is firmware or kernel or overlay but figured i'd start here! :)

I'd like to show people how to use older installs of raspbian on their Pi Zero, its been asked a few times and i expect it'll come up more soon. I noticed that if you replace just the firmware files on the card, it does boot firm/kernel just fine *but* you get
>  WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Device mode
if there's nothing plugged into the Zero's microUSB.
forever. seems like there's an OTG pin (which is super cool because of course you could have the zero act as a device) but its in the wrong state when no USB device is plugged in. So i guess my question is: if this OTG pin is floating, which makes the Zero think its a USB device, how does one set the pulldown/up, is this done in firmware, overlay, kernel...?

popcornmix, 2015-12-10:
You need this on Pi zero (if you don't have a correct USB OTG cable connected)
https://github.com/raspberrypi/linux/commit/59e76bb7e2936acd74938bb385f0884e34b91d72

ladyada, 2015-12-10:
oh got it, sneaky commit message ;)

closed 2015-12-10 by ladyada