raspberrypi/firmware #514: OMX_FreeHandle hangs video_decode with use of OMX_IndexParamBrcmExtraBuffers

jvcleave, 2015-12-11:
I am running into a situation where I am unable to use OMX_FreeHandle to free a handle to a video_decode component. This causes the RPi to hang or leak GPU memory.
My setup works like this

Create a tunneled application using

video_decode->image_fx->video_scheduler->clock/render (render can be video or egl)

The setup is similar to omxplayer and it's use of the image_fx tunnel for deinterlacing. The difference is that the user can change the filters while the application is running.

What works:
Run the application and provide a filter (e.g. Posterize) 
Shut it down via CTRL-C or CTRL-\

What doesn't work:
Run the application and provide a filter (e.g. Posterize) 
Change the filter (e.g. Posterize to Sketch, this part works but see Note* below)
Shutting it down via CTRL-C or CTRL-\ will start to close the application but will hang on OMX_FreeHandle with video_decode. The application will close if I omit this call but will leak GPU memory. All the other component handles are freed correctly.

*Note on changing filter:
I have to allocate an extra buffer via OMX_IndexParamBrcmExtraBuffers per "allowed changes". Meaning if I want to be able to go through 5 filters I need to allocate 5 extra buffers. If I don't do this the video_decode component will timeout.

I also have another application that uses the camera component with similar tunneling (substitute video_decode with the camera). This camera component doesn't have any issues switching filters or exiting with OMX_FreeHandle.

@popcornmix @julianscheel @6by9 Apologize for the tags but you come up fairly often when trying to search for related issues. Any insight would be appreciated!

jvcleave, 2015-12-11:
test file using a modified version of video.c and hello_videocube
https://gist.github.com/jvcleave/afb745c686b8456f5007

This doesn't hang on exit but does demonstrate the leak

6by9, 2015-12-11:
Which port did you use on the camera? The proprietary tunneling behaves differently between the preview port (70) and video encode port (71).
Does every effect leak a buffer? It sounds like a framework thing rather than the effects themselves, but would be useful to confirm.

jvcleave, 2015-12-11:
OMX_ALL actually (something video_decode doesn't accept)

What's the best way to check if for the leaked buffers? The commands I am using are

`sudo vcdbg pools`
This will display nothing if the handle is freed but content otherwise

and
`sudo vcdbg reloc | grep "free memory"`
This will get dramatically lower when FreeHandle fails

6by9, 2015-12-11:
OMX_ALL is not a valid port for tunneling to image_fx. There are some parameters that are set on OMX_ALL as they apply to the entire component, but that's a bit different.

IL buffers are likely to be allocated with the description "ilin mapping buffer", not that I've directly checked it. If you've directly tunneled video_decode to image_fx, I suspect you're leaking images from the decode image pool rather than actual IL buffers. When setting up a tunnel, if possible the components will negotiate to do proprietary tunnels, and you'll find the PortDefinition nBufferCountActual gets set to 0 as it doesn't use actual buffers (they swap function pointers between the components so images can be passed directly).

...
Just looked at your code. Are you calling OMX_FreeHandle after calling ilclient_cleanup_components? ilclient_cleanup_components will have already called it, so you're doing a double free.

jvcleave, 2015-12-11:
the OMX_FreeHandle stuff is #if 0ed - I was trying to do it manually but video.c doesn't understand the Component structure

Ok - sorry about the misinformation about the Camera app. I am using Tunnels in the app but not an image_fx component/tunnel. The filters are applied directly to the Camera component (using OMX_ALL)

6by9, 2015-12-11:
OK, just wanted to confirm on OMX_FreeHandle - I was expecting to see
```
#if 0
   ilclient_cleanup_components(...);
#else
   //OMX_FreeHandle stuff
#endif
```
At the moment if I converted the #if 0 to #if 1 I'd get a double free. Actually I'd get a triple free as the end of the #if 0 also appears to call ilclient_cleanup_components!

Filters applied directly by the camera go by a very different path, but that would tend to confirm that the plugins themselves are OK and it is in the image_fx component. 
Your original comment had said:
"I also have another application that uses the camera component with similar tunneling (substitute video_decode with the camera)." which would have included image_fx.

There is a parameter OMX_IndexParamBrcmDisableProprietaryTunnels that forces an output port to use standard buffers rather than proprietary tunnels. If you set that on video_decode's output port it would allow me to narrow down where the issue is a little more.

jvcleave, 2015-12-11:
Thanks - I just tried `OMX_IndexParamBrcmDisableProprietaryTunnels` on the gist app. It seems to be the same behavior (exits but leaks).

6by9, 2015-12-11:
Hmm, not the behaviour I was expecting. I'll add it to the queue of issues to check, unless @popcornmix can check it sooner.
This sounds like exactly the same pipe that Kodi and OMXPlayer use, so I'm slightly surprised there's an issue.

jvcleave, 2015-12-11:
Yeah - the main difference in omxplayer that it doesn't change the filters on the fly. 

I'm noticing something about the videocube test app - it's not making it to `ilclient_cleanup_components` so trying to clear that up now

6by9, 2015-12-11:
Ah, fair point. They only use the one filter (deinterlace) and drop the component in and out of the pipe.

Back at Brcm we did have a test app that ran a test source->image_fx->video_render, and switched through all the effects. I don't think the component has changed significantly since, so it's a little strange that it is broken now. 
I'll try to run up your test later in the weekend.

6by9, 2015-12-11:
One last thought, can you drop the deinterlacers out of the list please? Those again use a different code path (they require temporal data, so hold onto old frames), whereas the rest all just work on the current frame.

jvcleave, 2015-12-11:
Yeah - In the current state I am just using the first 4. Same behaviour

The current hang seems to happen with `ilclient_disable_port_buffers` on the video_decode input port (130)

Ruffio, 2016-07-24:
@6by9 have you had the time to test as stated 8 months ago? Any hope of doing this in near future?

6by9, 2016-07-24:
Not looked at it.
@popcornmix was having some similar oddballs off image_fx and deinterlacing recently via MMAL (common GPU code). I can't recall if we got to the bottom of all the issues, and I think some of his were due to more than 32 input buffers.

JamesH65, 2017-12-21:
@6by9 Status?

jvcleave, 2018-12-13:
Happy Anniversary ;)

Happened to be revisiting this but disabling proprietary tunnels on video_decode output port seems to stop the leaking. There seems to be a bit of a performance hit for this (may be expected)

```
OMX_PARAM_BRCMDISABLEPROPRIETARYTUNNELSTYPE tunnelConfig;
        OMX_INIT_STRUCTURE(tunnelConfig);
        tunnelConfig.nPortIndex = 131;
        tunnelConfig.bUseBuffers = OMX_TRUE;
        
        //performance hit but stops the GPU from leaking on exit
		if (OMX_SetParameter(ILC_GET_HANDLE(video_decode), OMX_IndexParamBrcmDisableProprietaryTunnels, &tunnelConfig) != OMX_ErrorNone)
       {
           printf("OMX_IndexParamBrcmDisableProprietaryTunnels failed.\n");
           exit(1);
       }
```

jvcleave, 2018-12-13:
To avoid the performance hit it appears I can just use the above code before exiting/deleting the encoder and that works as well.

closed 2018-12-16 by jvcleave

oomek, 2019-03-25:
Did anyone find a solution for this hanging on `ilclient_disable_port_buffers()` without sacrificing the performance by any chance?