raspberrypi/firmware #519: Unclear what license the *.dtb files are covered by in boot

skrll, 2015-12-19:
Please be explicit about *.dtb file licensing. Thanks.

popcornmix, 2015-12-19:
Do you have a preference?

I think we are happy to allow use under either GPLv2 or LICENCE.broadcom whichever is most convenient.
I'm temped to make this licence specify LICENCE.broadcom for all files in /boot (including kernel*.img and *.dtb).
However kernel*.img and *.dtb can also be built from source from the GPLv2 licensed linux tree, so your own builds of these files can use GPLv2 is that is more convenient.

@pelwell ?

pelwell, 2015-12-19:
I think they should be considered part of the kernel, with the same licensing, so GPLv2.

jaredmcneill, 2015-12-19:
Where is the source for the .dtb files, if they are to be considered GPLv2?

pelwell, 2015-12-19:
I don't have a problem with that, but the overlays have been contributed by several individuals who may have other wishes.

jaredmcneill, 2015-12-19:
Sure, I understand. Clarifying licensing of the files that come from Raspberry Pi Foundation would be a great start though!

popcornmix, 2015-12-19:
Okay, I've added the ``*.dtb`` files alongside the ``kernel*.img`` files as GPLv2.
Whilst we wouldn't have an objection to more permissive licensing, it is complicated by the multiple contributors, so sticking with GPLv2 is safest.

clivem, 2015-12-19:
Upstream preference is for dual GPL/X11 licensing for DT files, ISTR.

closed 2016-01-15 by popcornmix