raspberrypi/firmware #521: Nethogs fails to start

PeterPablo, 2015-12-29:
I updated my Raspberry Pi B using `sudo BRANCH=next rpi-update` to `Linux rpi 4.4.0-rc7+ #831 PREEMPT Mon Dec 28 19:09:59 GMT 2015 armv6l GNU/Linux`.

Nethogs fails to start:
```
root@rpi:~# nethogs eth0
creating socket failed while establishing local IP - are you root?
```

```
nethogs -V
 version 0.8.0
```

Related bug report:
https://bugzilla.redhat.com/show_bug.cgi?id=1294363

I am reporting here, since this works with the current stable firmware/kernel. Please close, if I should report this in a different place.

popcornmix, 2015-12-29:
Can you use the "report a bug option" here: https://www.raspbian.org/RaspbianBugs

PeterPablo, 2016-01-07:
Bug report created: https://bugs.launchpad.net/raspbian/+bug/1531847

closed 2016-01-07 by PeterPablo

edouard-lopez, 2016-01-28:
I got this issue on Kubuntu `15.10`

### Kernel
```
$ uname -a
Linux xyz 4.2.0-27-generic #32-Ubuntu SMP Fri Jan 22 04:49:08 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

### Release
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 15.10
Release:        15.10
Codename:       wily
```

### Version

```
$ nethogs -V
version 0.8.0
```

pelwell, 2016-01-28:
That's not a firmware bug, so this is the wrong forum. You aren't using our distribution or even our kernel, so this is completely the wrong forum.

edouard-lopez, 2016-01-28:
@pelwell sorry didn't pay attention, too early in the morning I guess :/