raspberrypi/firmware #522: vchiq doesn't release PWM

mlelstv, 2015-12-31:
I'm using the PWM to control external hardware. This of course conflicts with the use
of the PWM for audio output, especially when done through the vchiq service. Unfortunately
the DMA channel used by the GPU remains active even when the vchiq service is closed.

Is there a way to make the vchiq service release the PWM?

popcornmix, 2015-12-31:
Does ``force_pwm_open=0`` added to config.txt help?

mlelstv, 2015-12-31:
On Thu, Dec 31, 2015 at 03:23:27AM -0800, popcornmix wrote:

> Does ``force_pwm_open=0`` added to config.txt help?

Thanks. This does help.

after boot:
GPU 3_CS:        20000008
GPU 3_CONBLK_AD: 00000000
GPU 3_DEBUG:     04018300

after vchiq closed and force_pwm_open=1 (or whatever the default is)
GPU 3_CS:        20F10021
GPU 3_CONBLK_AD: FE4B8380
GPU 3_DEBUG:     04088300

after vchiq closed and force_pwm_open=0
GPU 3_CS:        20F10001
GPU 3_CONBLK_AD: 00000000
GPU 3_DEBUG:     04018300

What is the reason to keep the DMA channel running?

Greetings,
-- 
                                Michael van Elst
Internet: mlelstv@serpens.de
                                "A potential Snark may lurk in every tree."

popcornmix, 2016-01-01:
Closing and opening PWM causes a pop sound, so by default we leave it open after first use. The ``force_pwm_open=0`` option disables this.

popcornmix, 2016-01-09:
Okay to close?

mlelstv, 2016-01-10:
On Sat, Jan 09, 2016 at 06:46:15AM -0800, popcornmix wrote:
> Okay to close?

Sure, the problem is understood. If I find something better to
avoid the pops, we can continue :)

-- 
                                Michael van Elst
Internet: mlelstv@serpens.de
                                "A potential Snark may lurk in every tree."

closed 2016-01-15 by popcornmix