raspberrypi/firmware #523: h264: Can't hardware accelerated play

Yuri666, 2016-01-03:
One sat channel (sd, h264) doesn't play with hardware acceleration in kodi - i see black screen.
If hardware acceleration disabled, image and sound appears, but with glitches.
In x86 linux all play fine.
Simple ts: https://yadi.sk/d/fGGfU_W9mePrW

popcornmix, 2016-01-03:
What version of kodi?
Does disabling omxplayer (but leaving mmal enabled) work okay?

Yuri666, 2016-01-03:
version of кodi does not depend - i tryed with 14, 15.2 and 16beta2
with disabled omxplayer and enabled mmal - no image but have sound (periodically glitched)

popcornmix, 2016-01-03:
It's playing for me with MMAL (but not omxplayer) with a [nightly/milhouse](http://forum.kodi.tv/showthread.php?tid=250817) build, so it's probably a kodi rather than a firmware issue.

There is a problem with the file:
```
04:13:32 73165.250000 T:1796674576   ERROR: ffmpeg[6B171410]: [h264] non-existing PPS 0 referenced
04:13:32 73165.250000 T:1796674576   ERROR: Previous line repeats 1 times.
04:13:32 73165.250000 T:1796674576   ERROR: ffmpeg[6B171410]: [h264] decode_slice_header error
04:13:32 73165.250000 T:1796674576   ERROR: ffmpeg[6B171410]: [h264] no frame!
04:13:32 73165.250000 T:1796674576   ERROR: ffmpeg[6B171410]: [h264] non-existing PPS 0 referenced
...
04:13:32 73165.273438 T:1796674576   ERROR: ffmpeg[6B171410]: [h264] decode_slice_header error
04:13:32 73165.273438 T:1796674576   ERROR: ffmpeg[6B171410]: [h264] no frame!
04:13:32 73165.312500 T:1796674576   ERROR: ffmpeg[6B171410]: [h264] mmco: unref short failure
04:13:32 73165.312500 T:1796674576   ERROR: Previous line repeats 1 times.
04:13:32 73165.312500 T:1796674576   ERROR: ffmpeg[6B171410]: [h264] number of reference frames (0+4) exceeds max (3; probably corrupt input), discarding one
04:13:32 73165.351562 T:1796674576   DEBUG: ffmpeg[6B171410]: [mpegts] PES packet size mismatch
```

which is presumably causing the issue, but it looks like it is possible to play.

Yuri666, 2016-01-03:
ok, i will try kodi 17, but in vdr i have same issue.

Yuri666, 2016-01-03:
in my opinion it's referenced with issue #499

popcornmix, 2016-01-03:
No, it's not that.
Can you create an advancedsetting.xml file containing:
```
<advancedsettings>
    <omx>
        <omxdecodestartwithvalidframe>0</omxdecodestartwithvalidframe>
    </omx>
</advancedsettings>
```

Yuri666, 2016-01-03:
I mean what both issues gone when disable hw acceleration.
I created file, but hasn't visible effect (kodi 15.2)

popcornmix, 2016-01-03:
In ``~/.kodi/userdata/advancedsettings.xml``?
Can you post a debug log so I can confirm that file is being read correctly?

Yuri666, 2016-01-03:
ok, advancedsettings, not advancedsetting as in previous message. :)
All work fine! Thank you!
Issue can be closed.

popcornmix, 2016-01-03:
Okay. The problem with the file is no IDR frames and no SEI recovery points.
Keep an eye here: https://github.com/raspberrypi/firmware/issues/375
which is a similar issue.

closed 2016-01-03 by popcornmix