raspberrypi/firmware #525: Incorrect display if MMAL_DISPLAYREGION_T.pixel_x / pixel_y are too high

wm4, 2016-01-05:
If both pixel_x and pixel_y are set to 2822400, then it appears the video is stretched to display size. It works when limiting the maximum value. (I tried setting both to 16000, which appeared to work.)

Some sort of overflow or overflow check in the firmware? It also ignores the bounds set by MMAL_DISPLAY_SET_DEST_RECT, so something strange must be going on here. (I'm not setting the ``mode`` or ``noaspect`` fields.)

This is trivial to work around (I'll just limit both fields to 16000), but I thought I'd report it.

popcornmix, 2016-01-07:
I've added a possible fix - can you test:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_pixel.zip

wm4, 2016-01-07:
Thanks. Will test next time I get on my RPI, hopefully tomorrow.

wm4, 2016-01-08:
Doesn't seem to work. On the other hand, maybe I failed to install the firmware correctly? After copying the contents of the zip to ``/boot/`` (overwriting the existing files) on my Raspbian and rebooting, I get:
```
$ /opt/vc/bin/vcgencmd version
Nov  1 2015 14:47:45 
Copyright (c) 2012 Broadcom
version 8f1c8e72174c0415fd479c1c7be66b880fab3d79 (clean) (release)
```

For my test, I've set pixel_x and pixel_y both to ``1<<27``.

popcornmix, 2016-01-08:
Do you have a test app I can run that does this?

wm4, 2016-01-08:
No, I just used mpv. Here is where it's set: https://github.com/mpv-player/mpv/blob/master/video/out/vo_rpi.c#L191

popcornmix, 2016-01-08:
Do you have a pre-built mpv with pixel_x and pixel_y both set to ``1<<27``that I can run on raspbian?

wm4, 2016-01-08:
You could try this: https://0x0.st/oRp.gz

Not sure if it will work. The libasound is because the mpv binary links against the channel map API, which is not available in Raspbian's libasound (at least not on my aging installation). I've tested it by playing a 720p h264 file with ``mpv file.mkv`` on a 1680x1050 screen. In my case, it actually doesn't render a video at all, and just shows a green line of some sort. Changing panscan with the w and e keys changes the video rectangle and can have interesting effects.

popcornmix, 2016-01-08:
``./mpv: error while loading shared libraries: libavfilter.so.6: cannot open shared object file: No such file or directory``

wm4, 2016-01-08:
libavfilter without the other stuff: https://0x0.st/oRf.gz

popcornmix, 2016-01-08:
It's running, but view is correct (with all panscan settings)
```
pi@domnfs:~/tests/mpd $ tvservice  -s
state 0x12000a [HDMI DMT (57) RGB full 16:10], 1680x1050 @ 60.00Hz, progressive
pi@domnfs:~/tests/mpd $ LD_LIBRARY_PATH=. ./mpv ~/dell/Doctor.Who.2005.S08E06.The.Caretaker.HD.TV.mkv 
Playing: /home/pi/dell/Doctor.Who.2005.S08E06.The.Caretaker.HD.TV.mkv
 (+) Video --vid=1 (*) (h264)
 (+) Audio --aid=1 (*) (aac)
flushdec
ALSA lib conf.c:3718:(snd_config_update_r) Cannot access file /usr/local/share/alsa/alsa.conf
ALSA lib pcm.c:2267:(snd_pcm_open_noupdate) Unknown PCM default
[ao/alsa] Playback open error: No such file or directory
[ao/oss] Can't open audio device /dev/dsp: No such file or directory
[ao] Failed to initialize audio driver 'oss'
Could not open/initialize audio device -> no sound.
Audio: no audio
Using hardware decoding (rpi).
VO: [rpi] 1280x720 mmal
flushdec:06 / 00:45:11 (0%) Cache: 10s+73MB
```

popcornmix, 2016-01-08:
Firmware gets
```
pixel_x = 134217728 (0x8000000)
pixel_y = 134217728 (0x8000000)
```
and the patch I added (in the custom firmware) is adjusting that correctly:
```
pixel_x = 32768 (0x8000)
pixel_y = 32768 (0x8000)
```
and avoids the overflow.

popcornmix, 2016-01-08:
Looks like you are not using the updated firmware.
Are you specifying a non-default start file in config.txt?
Are you using NOOBS or another distribution with multiple "boot" directories?

wm4, 2016-01-08:
Did I fail to install your firmware package correctly?

popcornmix, 2016-01-08:
Yes - the date reported should be yesterday. What does
``md5sum /boot/start* ``
report?

wm4, 2016-01-08:
```
950d8748d1b44e2a015535f0c80ec0e8  /boot/start_cd.elf
77b4d3b1d75827445436765541c11852  /boot/start_db.elf
911c2f60fd865423f926be9ecd392340  /boot/start.elf
158dac9824b00505fc3cf884ef01c708  /boot/start_x.elf
```

popcornmix, 2016-01-08:
```950d8748d1b44e2a015535f0c80ec0e8  start_cd.elf
911c2f60fd865423f926be9ecd392340  start.elf
158dac9824b00505fc3cf884ef01c708  start_x.elf
```
You don't have ``start_debug=1`` in config.txt? You have rebooted since updating?

wm4, 2016-01-08:
>You don't have ``start_debug=1`` in config.txt?

I have this in config.txt. Should I comment it out?

>You have rebooted since updating?

Yes.

popcornmix, 2016-01-08:
Yes, comment out start_debug=1. It uses the ``start_db.elf`` which doesn't get built in test firmware.

wm4, 2016-01-08:
It appears this was the problem.

It works now.

popcornmix, 2016-01-08:
Okay - the fix will be in next firmware pushed.
The start_debug is rarely used, so I don't normally build it for test firmwares (saves a bit of building time), but I'll consider building it by default.

wm4, 2016-01-08:
Thanks. I'm a bit confused about start_debug. What exactly does it do? Just switch which firmware image to boot?

popcornmix, 2016-01-08:
It chooses a debug (in logging rather than build flags) enabled firmware.
It is optional because it is larger (by about 1MB) and slower. For GL heavy applications it can be considerably slower.

If you are using the firmware and having errors, then switching on ``start_debug=1`` allows ``sudo vcdbg log assert`` to provide more messages about when things fail.
However you should probably disable it when you are not debugging firmware side errors.

popcornmix, 2016-01-13:
This is now in rpi-update firmware. Okay to close?

wm4, 2016-01-13:
Of course.

closed 2016-01-13 by wm4