raspberrypi/firmware #526: a standard /bin/dt-blob.bin stops pi from booting up since Oct 21, 2015 release

hellochenwang, 2016-01-10:
dt-blob.bin is compiled from https://www.raspberrypi.org/documentation/configuration/images/dt-blob.dts (https://www.raspberrypi.org/documentation/configuration/pin-configuration.md)

prior to commit #07e5170cee4c83a5c34b3f73c6497c5e88be3484, this dt-blob.bin file works fine.
since that commit, pi2b2 won't boot unless i remove /boot/dt-blob.bin

ref:https://github.com/Hexxeh/rpi-firmware/commit/07e5170cee4c83a5c34b3f73c6497c5e88be3484#commitcomment-15359496

6by9, 2016-01-10:
@popcornmix Sorry, saw this a week or so back, but forgot to let you know.
It's missing DISPLAY_I2C_PORT, and if not defined we get GPIOMAN_PIN_INVALID back which is -1. That is used in to the display driver unchecked - opening I2C port -1 falls in a heap.

P33M, 2016-01-11:
git blame == me.

I can fix the regression introduced in https://github.com/Hexxeh/rpi-firmware/commit/07e5170cee4c83a5c34b3f73c6497c5e88be3484 which added support for DSI displays on Model A and B, however this will mean the dt-blob.dts in the documentation will cause display probing to fail (but not a crash any more) as DISPLAY_I2C_PORT needs to be present and valid to probe for the display.

The documentation dt-blob.dts needs updating.

popcornmix, 2016-01-11:
I've got a fix for unset DISPLAY_I2C_PORT not hanging, and we're planning to add dt-blob.dts to firmware repo (and documentation needs to point to that).

popcornmix, 2016-01-13:
Latest firmware should no longer hang with your current dt-blob.dtb file.
We have also added an up to date blob file to repo:
https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts
which should also avoid the problem.

popcornmix, 2016-01-24:
@hellochenwang okay to close?

closed 2016-03-21 by popcornmix