raspberrypi/firmware #527: dtoverlay merge fails for large overlays

fabled, 2016-01-13:
I'm building add-on board with several peripherals and dt overlay size is about 7kB. It seems that after certain limit the bootloader fails to merge in the overlay. This seems to be purely size limit, as each individual block of the overlay work in any combination, and I'm also able to make working dtb with all attached peripherals by just leaving few gpio-key mappings out.

Kernel is configured with OF_OVERLAY=n.

The bootloader dtdebug prints the following:
```
002456.375: dtoverlay_merge_params failed - err -3
```

Full dts, dtb and dtdebug available if it helps.

pelwell, 2016-01-13:
There is a limit of 0x3f00 bytes for the combined .dtb (i.e. base dtb + overlays). This is because the .dtb is loaded at 0x100 and, although the kernel is loaded at 0x8000, something fails if we write into the area 0x4000-0x7fff. You can override the load position using the config.txt variables device_tree_address (default 0x100) and device_tree_end (default 0x4000), but you will probably need to change kernel_address (default 0x8000) at the same time, and that may have consequences.

I had thought that 16KB would be sufficient, but if that isn't the case then a rethink is needed.

fabled, 2016-01-13:
0x3f00 for all dtb data seems to be not enough for me. The result we get is more than that.

So far I had all load addresses at their defaults with the exception of loading initramfs to 0x08000000.

I tried changing dtb area elsewhere, and moving dtb area end + kernel start; but that seemed to result in boot failure. Do you have some suggested addresses I could try? Or does this require additional changes to kernel config or similar?

pelwell, 2016-01-13:
It's probably going to be better all round if I tackle the problem at source and remove the restriction. Can you upload your files somewhere (dropbox, etc.) as test cases.

In the meantime, as a workaround you could create a custom base .dtb by deleting the things you don't need - the aliases, for example.

fabled, 2016-01-13:
The overlay sources are at:
https://github.com/rameplayerorg/rameplayer-alpine/tree/master/dts

See rame-overlay.dts that is basically merge of all the other overlays. There's attached i2c gpio expander, several gpio-keys, gpio-leds and gpio rotary-encoder mappings as well as add-on audio board and rtc clock. :)

pelwell, 2016-01-13:
Thanks. I would encourage you to try the workaround - I'm not sure when I'll get round to this.

pelwell, 2016-01-14:
I've got a firmware patch to allow arbitrarily large DTBs. With it applied, your overlay loads (although the rotary-encoder driver warns about "invalid GPIO -512").

Assuming it passes testing it should appear in an rpi-update release soon.

fabled, 2016-01-14:
Very nice. I'll be happy to test it. - And yes, the dtb works fully only on patched kernel. The rotary-encoder and gpio-mcp23s08 have few fixes that I'm working on to be upstreamed, and the audio add on board needs the cirrus patch.

pelwell, 2016-01-15:
A firmware update including this patch has been pushed to Hexxeh's rpi-firmware repo. rpi-update to install it, and remove any device_tree_address, device_tree_end or kernel_address settings from config.txt.

fabled, 2016-01-15:
It seems to not work for me :/

I still get:
```
002317.646: dtoverlay_merge_params failed - err -3
```
This is on rpi2 (the largest base dtb). No addresses specified, except that initramfs is loaded; I tried both with initramfs address 0x08000000 and 'followkernel' and the result was same.

And kernel dmesg says:
```
[    0.254845] raspberrypi-firmware raspberrypi-firmware.0: Attached to firmware from 2016-01-15 15:56
```
so the firmware image should be correct. Though I have only start.elf on the boot media. But my understanding is that the other start*.elf are not needed under normal circumstances.

pelwell, 2016-01-15:
It looks like there was some finger trouble in the firmware production - none of the ELFs have been updated. Check back in about 15 minutes and there should be another release, this time incorporating this change.

pelwell, 2016-01-15:
Try now - it works for me.

fabled, 2016-01-15:
Yes! Thank you!

vcdbg says now:
```
002566.103: Device tree loaded to 2efebf00 (size 40c3)
```

and all looks green!

pelwell, 2016-01-18:
OK to close?

fabled, 2016-01-18:
yes.

closed 2016-01-18 by fabled