raspberrypi/firmware #528: mpeg2 video not playing properly, beginning with kernel 4.0.9

kiwi31415, 2016-01-15:
Hi,

I'm using raspbian wheezy with kodi as media center on two raspberry pis (B+ and 2). All raspbian packages are up to date.

Up to kernel version 4.0.8 (commit be02a7f5b5c560b6f33bddcd36b207b85b1d95a5) I'm able to watch DVB-T and MPEG2 Videos in HD (including mpeg2 key) perfectly.

Beginning with version 4.0.9 (commit 46d179597370c5145c7452796acbee0f1ff93392) videos don't play smooth anymore, the playback is very (!) jerky. The same applies to recorded TV-shows.

As a workaround I'm currently downgrading the kernel by running
rpi-update be02a7f5b5c560b6f33bddcd36b207b85b1d95a5

The situation is the same on both machines after a fresh install. There is also no change when using an almost empty config.txt (only containing the mpeg decoder key). It is also independent on the network or storage used.

Has there been some critical change between 4.0.8 and 4.0.9?

Please find a 1min example mpeg2 file here: 
https://mega.nz/#!iAF2iZBA!1HyUqSKvLOLDAnSVqi-ieiwjBrimaV5RFlmEbc3PQUU

Thanks for your help,

Thomas

popcornmix, 2016-01-15:
I imagine it is:

> firmware: video_decode: Don't wait for a valid timestamp to output frames

Let me see if I can find a way of achieving the effect of that commit without harming your file.

popcornmix, 2016-01-19:
The file seems to be playing well here.
I suspect the problem is the old version of Kodi. What version of Kodi are you using?

If you update to jessie you will get Kodi 15.2 in the repo and I don't think that has the issue.
You could also try OpenELEC or OSMC which I don't believe will have the issue.

kiwi31415, 2016-01-20:
I can confirm, the video plays smooth after updating to raspbian-jessie using the latest firmware. 

Hence, the problem really seems to be related to raspbian-wheezy and not to the kernel/firmware.

Thank you very much,

Thomas

closed 2016-01-20 by kiwi31415