raspberrypi/firmware #529: h264: tv stream playback failure

glenvt18, 2016-01-17:
This  [sample](https://drive.google.com/open?id=0BxEZpdTX1bPvRzRfN1FhYVVXa2s), recorded from TV fails to playback with openelec 6.0 and omxplayer - no video and audio (and no video with dvdplayer). It plays fine with ffmpeg (sw decoding).
[kodi.log](http://pastebin.com/j0cw9ku6)
[dmesg](http://pastebin.com/tQPcW8Ff)

popcornmix, 2016-01-17:
The sample actually plays with master build of Kodi and MMAL acceleration, so it seems the GPU can play the file. I get no picture with omxplayer.

I think the stream has no IDR frames and no valid SEI recovery points. Try adding an advancedsettings.xml file with:

```
<advancedsettings>
    <omx>
        <omxdecodestartwithvalidframe>0</omxdecodestartwithvalidframe>
    </omx>
</advancedsettings>
```

I think this is probably a duplicate of: https://github.com/raspberrypi/firmware/issues/375

glenvt18, 2016-01-17:
Yes, it looks like. I tried the sample from #375 and the symptoms are the same. Setting `omxdecodestartwithvalidframe` solves the issue. Thanks.

popcornmix, 2016-01-17:
Okay, I'll close this, but you may want to follow #375 in case a fix means omxdecodestartwithvalidframe is no longer required.

closed 2016-01-17 by popcornmix