raspberrypi/firmware #530: cannot boot rpi2 with latest firmware

jeffersonchua, 2016-01-22:
git pulled yesterday and can't boot rpi2 anymore. No display, no ethernet. Nothing.

The latest commit is ...
  commit 1ed8ecaa87ff7c925fe37be5bc557f5d9e4c29bd
  Author: popcornmix <popcornmix@gmail.com>
  Date:   Tue Jan 19 18:47:08 2016 +0000

Reverting back to start.elf, start_x.elf, fixup.dat, fixup_x.dat from the prior commit (Jan 07)  made the rpi2 boot up fine.

Same kernel used for both firmwares.
  commit 2c86f0a88213829d3a5ac9a8f2ad52335f8502bc
  Author: Phil Elwell <phil@raspberrypi.org>
  Date:   Tue Jan 19 17:16:38 2016 +0000

Anything new to set in the kernel? Or bad firmware?

Thanks,
Jeff.

popcornmix, 2016-01-22:
https://github.com/raspberrypi/firmware/commit/1ed8ecaa87ff7c925fe37be5bc557f5d9e4c29bd didn't change any of start.elf, start_x.elf, fixup.dat, fixup_x.dat. So reverting to previous firmware (which is Jan 15) shouldn't have an effect.

Can you link to the exact start.elf, start_x.elf, fixup.dat, fixup_x.dat commit you used?

MrEngman, 2016-01-22:
`
The latest commit is ...
 commit  1ed8eca 
 Author: popcornmix popcornmix@gmail.com
 Date: Tue Jan 19 18:47:08 2016 +0000
`

Is the latest firmware commit for rpi-4.1.y

`
And Same kernel used for both firmwares.
 commit 2c86f0a88213829d3a5ac9a8f2ad52335f8502bc
 Author: Phil Elwell phil@raspberrypi.org
 Date: Tue Jan 19 17:16:38 2016 +0000`

Is kernel source for rpi-4.4.y

Maybe that's why things are screwed up.

jeffersonchua, 2016-01-22:
May be I should further clarify ...

1) commit 1ed8eca is the latest "firmware" pulled. I did a fresh "clone"

2) commit 2c86f0a88213829d3a5ac9a8f2ad52335f8502bc is the latest "kernel" pulled. Compiled kernel for pi2.

I had a previous version of the (start.elf, start_x.elf, fixup.dat, fixup_x.dat) and that worked. The latest clone of (start.elf, start_x.elf, fixup.dat, fixup_x.dat) didn't work.

Here're the md5sum

Older working files from Jan 07 firmware clone ..
72e3a005f6d269d1929e322d194578a9  fixup.dat
09c523792aa6f2f839931d42e3b6e124  fixup_x.dat
1e84fa84fcb68e7d4a3ef62056dd17e7  start.elf
924615dbc50420d665695f21c9ffda64  start_x.elf

Newer non-working files from latest firmware clone ...
8a05a3effa683195f2b1cf67fb6f5879  fixup.dat
a026ab07a27ce260321959f6739a1d87  fixup_x.dat
bd03da327b01108c60f09cc4113b6f5f  start.elf
4af69fb07ed99f0e39c4dc75b0e245a2  start_x.elf

Thanks,
Jeff.

pelwell, 2016-01-22:
Can I confirm:
1) This is a 4.4 kernel you have compiled yourself? Did you use the bcm2709_defconfig?

2) This self-compiled kernel works with the older firmware but not the newer firmware?

Have you tried `sudo rpi-update BRANCH=next`? That is currently a top-of-tree 4.4 kernel.

jeffersonchua, 2016-01-22:
> Can I confirm:
> 1) This is a 4.4 kernel you have compiled yourself? Did you use the
> bcm2709_defconfig?
>
>  Yes. Compiled and running with patches pulled every week.

> Did you use the bcm2709_defconfig?

Started off with one, but have not gone back to use it for a while. That
could be the problem?

> 2) This self-compiled kernel works with the older firmware but not the
> newer firmware?
>
> Yes.

> Have you tried sudo rpi-update BRANCH=next? That is currently a
> top-of-tree 4.4 kernel.
Not yet. I'll try with reseting back to bm2709_defconfig and see if that
works. Or may be there's something new in there that I missed.

Thanks,
Jeff

jeffersonchua, 2016-01-25:
On Sat, Jan 23, 2016 at 6:24 AM, Jeff Chua <jeff.chua.linux@gmail.com>
wrote:
> On Sat, Jan 23, 2016 at 12:34 AM, Phil Elwell <notifications@github.com>
> wrote:
>
>
>
> Started off with one, but have not gone back to use it for a while. That
> could be the problem?
>
>
> Not yet. I'll try [...]

Started off with a new bm2709_defconfig and this time the newer firmware
works. I traced down this difference ...

--- .config.org 2016-01-23 21:43:14.000000000 +0800
+++ .config     2016-01-25 07:56:11.000000000 +0800
@@ -377,11 +377,11 @@
 CONFIG_HAVE_ARM_ARCH_TIMER=y
 # CONFIG_MCPM is not set
 # CONFIG_BIG_LITTLE is not set
-CONFIG_VMSPLIT_3G=y
+# CONFIG_VMSPLIT_3G is not set
 # CONFIG_VMSPLIT_3G_OPT is not set
-# CONFIG_VMSPLIT_2G is not set
+CONFIG_VMSPLIT_2G=y
 # CONFIG_VMSPLIT_1G is not set
-CONFIG_PAGE_OFFSET=0xC0000000
+CONFIG_PAGE_OFFSET=0x80000000
 CONFIG_NR_CPUS=4
 CONFIG_HOTPLUG_CPU=y
 CONFIG_ARM_PSCI=y

It seems the newer firmware now expect the kernel to be using 2G/2G VM
Split or Page Offset at 0x80000000 instead of 0xC0000000 ?

Jeff

closed 2016-03-21 by popcornmix