raspberrypi/firmware #532: Running EGL applications in a namescaped process hierarchy fails

agherzan, 2016-01-29:
We have a simple EGL & OMX application which runs fine on the host. When running the same binary in a container, the application behaves strangely (lacks transitions and other animation components) while "vcdbg log msg" complains with:

`025819.942: *** No KHAN handle found for pid 24
025836.615: *** No KHAN handle found for pid 24
025853.285: *** No KHAN handle found for pid 24`

Running the container without a new process namespace, everything gets back to normal. Our theory is that the kernel doesn't do the process id conversion in the namespace context. 

Can you guys confirm this issue? Are you able to provide any additional information which would fix our issue?

popcornmix, 2016-01-30:
Can you explain exactly what you did to run in a container?
At a guess I'd suspect something different is happening with:
https://github.com/raspberrypi/userland/blob/master/interface/vcos/pthreads/vcos_platform.h#L622

pelwell, 2016-01-30:
I think the Khronos APIs use a VCHIQ feature that tells the server the client's PID. With a bit more information it may be that a fix is achievable.

agherzan, 2016-02-01:
I will open an issue in linux-raspberrypi cause I have a fix for this and it applies to that repository,

agherzan, 2016-02-01:
https://github.com/raspberrypi/linux/pull/1279

Ruffio, 2016-07-29:
@agherzan I guess this issue can be closed?

agherzan, 2016-08-02:
Indeed it can. Thanks.

closed 2016-08-02 by agherzan