raspberrypi/firmware #534: Proper cleanup of resources in hello_triangle

carlonluca, 2016-02-02:
For an open project I'm implementing a code that creates and destroys EGL surfaces and contexts to create OpenGL textures in background threads. Everything works properly until this creation/destruction is iterated for days (which means thousands of times). In that case I get many types of errors, but typically EGL_BAD_PARAMETER from EGL. After that I need to restart my app. I therefore tried to reproduce with the hello_triangle example, and placed the main function (which I see includes the cleanup procedure) in a loop. After a thousand iterations it fails. This time the error seems to come from somewhere else.

As this example does something pretty similar to what I do, would it be possible for you to check why it is not looping forever? I suppose if everything is properly disposed it should loop endlessly. My hope is that making this work may help me fix my code.

The code I'm attaching here is similar, it just adds glDeleteTextures for the texture IDs, a couple of calls that I tried to place to close the dispmanx window and the bcm_host_deinit() to compensate the bcm_host_init() call. Unfortunately still it fails creating the EGL surface after exactly 1024 iterations with the error EGL_BAD_NATIVE_WINDOW, so I guess something is still not released properly. Any idea what?

Thank you!

[triangle.c.zip](https://github.com/raspberrypi/firmware/files/114316/triangle.c.zip)

popcornmix, 2016-02-02:
I can see a few bugs in exit_func.
You have cleanup code inside asserts. That will disappear in a release build.
All gl/egl operations need a valid context. You are destroying the context before deleting textures/surfaces.
You need to bracket vc_dispmanx_element_remove inside a vc_dispmanx_update_start/vc_dispmanx_update_submit_sync pair or no operation will occur (dispman operations are just queued up and only executed on the submit).

Here is my suggested exit_func
```
static void exit_func(void)
// Function to be passed to atexit().
{
   int s;
   // clear screen
   glClear( GL_COLOR_BUFFER_BIT );
   eglSwapBuffers(state->display, state->surface);

   glDeleteTextures(6, state->tex);
   eglDestroySurface( state->display, state->surface );

   dispman_update = vc_dispmanx_update_start( 0 );
   s = vc_dispmanx_element_remove(dispman_update, dispman_element);
   assert(s == 0);
   vc_dispmanx_update_submit_sync( dispman_update );
   s = vc_dispmanx_display_close(dispman_display);
   assert (s == 0);

   // Release OpenGL resources
   eglMakeCurrent( state->display, EGL_NO_SURFACE, EGL_NO_SURFACE, EGL_NO_CONTEXT );
   eglDestroyContext( state->display, state->context );
   eglTerminate( state->display );

   // release texture buffers
   free(state->tex_buf1);
   free(state->tex_buf2);
   free(state->tex_buf3);

   //printf("\ncube closed\n");
} // exit_func()
```

carlonluca, 2016-02-06:
The example now works perfectly and thanks to your observations I could fix my code. I can reach more than 120,000 allocations/deallocations without any crash. Can you fix the example code or do you accept a pull request?
Thank you!

popcornmix, 2016-02-06:
I've updated locally (http://paste.ubuntu.com/14913697/)
Will push with next firmware update (when I've tested it).

popcornmix, 2016-02-20:
This has been pushed. Okay to close?

LucaCarlon, 2016-02-20:
Yes of course. Everything seems to be working properly.

carlonluca, 2016-02-20:
Yes of course. Everything seems to be working properly.

closed 2016-02-20 by popcornmix