raspberrypi/firmware #535: MMAL decoder sometimes returns broken timestamp instead of MMAL_TIME_UNKNOWN

wm4, 2016-02-03:
Trying to decode mpeg-2, which often has frames with no PTS set, e.g. AV_NOPTS_VALUE in FFmpeg. On the RPI, MMAL_TIME_UNKNOWN indicates the lack of a timestamp. I pass my timestamps through as double reinterpret-casted to int64_t, so the timestamps might look strange to MMAL, though "no timestamp" is still passed as MMAL_TIME_UNKNOWN. At one point, instead of a missing timestamp, I get a seemingly random value back, which was never sent to MMAL, and which isn't equal to MMAL_TIME_UNKNOWN. An example is 10176886159610705 at the end of the following list. The expected value is MMAL_TIME_UNKNOWN.

```
dts_in=-9223372036854775808 
pts_in=-9223372036854775808 
dts_in=-4638402524338454288 
pts_in=-4713475920839573504 
dts_in=-4713475920839573504 
pts_in=-4332462841530417152 
dts_in=4584969416439529456 
pts_in=4584969416439529456 
dts_in=4590674800626663856 
pts_in=-4332462841530417152 
dts_in=4593079122347729392 
pts_in=4593079122347729392 
dts_in=4595178412263633364 
pts_in=4595178412263633364 
dts_in=4596380573124166128 
pts_in=-4332462841530417152 
dts_in=4598179416900129126 
pts_in=4598179416900129126 
dts_in=4598780497330395508 
pts_in=4598780497330395508 
dts_in=4599682218055786800 
pts_in=-4332462841530417152 
dts_in=4600283298486053182 
pts_in=4600283298486053182 
dts_in=4601185019211444476 
pts_in=4601185019211444476 
dts_in=4601786099641710858 
pts_in=-4332462841530417152 
dts_in=4602683219689882811 
pts_in=-4332462841530417152 
dts_in=4602983759905016003 
pts_in=4602983759905016003 
dts_in=4603434520187719929 
pts_in=-4332462841530417152 
dts_out=-9223372036854775808 
pts_out=4584969416439529456 
dts_in=4603735060402853121 
pts_in=4603735060402853121 
dts_out=-9223372036854775808 
pts_out=10176886159610705 
```

MMAL_PARAMETER_VIDEO_INTERPOLATE_TIMESTAMPS is disabled.

This doesn't happen if I pass timestamps in millisecond units. Is there maybe some sort of overflow somewhere?

I can just switch to integer timestamps, but I thought I'd report it anyway.

wm4, 2016-02-03:
Also, what is the correct way to retrieve the PTS? The output buffer DTS is always MMAL_TIME_UNKNOWN. (I'd expect a reordered DTS, which could be used to fill out the PTS for such mpeg streams. FFmpeg does this.)

6by9, 2016-02-04:
Pass on why you get seemingly random timestamps. They should follow through that input PTS is added to output PTS unless manipulation is necessary (eg deinterlacing).
AIUI In video_decode the timestamp and other flags get added to a queue so they can be picked up on the other side of the codec. I would have expected nothing to fiddle with them if INTERPOLATE_TIMESTAMPS wasn't set, but I haven't checked.

> Also, what is the correct way to retrieve the PTS? The output buffer DTS is always MMAL_TIME_UNKNOWN.

Any MMAL component that starts "vc.ril." is actually sitting on top of an IL component on VideoCore. IL only has one timestamp field on buffers, and that is used for PTS. As above, they should just follow from input buffer to output.

6by9, 2016-02-04:
Simple answer is don't try to do clever things with timestamps being fed in if there is any chance of feeding a MMAL_TIME_UNKNOWN as a PTS.

Snippet of code from the firmware source:
```
         // sanity check the timestamps for streams with B frames as we
         // can be fed the wrong pts in that case
         if(id->has_b_frames && dec->info.frame_type != FRAME_TYPE_B && id->valid_next_frame_time &&
            dec->timestamp < id->last_frame_time)
         {
            // something is wrong with the timestamp. Try to correct it.
            OMX_TICKS timestamp = (id->next_frame_time+500)/1000;
```

And whilst I was right in saying that IL only has a single timestamp on buffers, MMAL will take in order:
- PTS if not MMAL_TIME_UNKNOWN
- DTS if not MMAL_TIME_UNKNOWN, and add a buffer flag to say so.
- Set to 0.
Once the IL side has seen one buffer that is viewed as having a DTS instead of PTS, it will assume all buffers have DTS until it is flushed.

I don't know decode well enough to comment on whether things are sane or not. Someone has developed a load of code to try to cope with bizarre streams, but it sounds like you have some even more bizarre ones.

wm4, 2016-02-05:
My absurd timestamps come from the fact that I was reinterpret-casting floats to integer (for the purpose of passing them through losslessy).

Since it makes no sense to interpret these timestamps as microseconds, this firmware code probably gets triggered accidentally, and there might be no problem with the sample file itself. I can still upload it, if you want. (I'd have to find out which file it was.)

Anyway, I've changed to feeding the decoder with actual microsecond timestamps now. It's still a potential problem for other ffmpeg API users. The timestamps are 64 bit integers, but treated as opaque, so the user is free to send anything. Possibly I could rescale the timestamps to microseconds if the API user sets a timebase...

JamesH65, 2017-12-21:
@wm4 @6by9 Reading through comments, I suspect this can now be closed?

6by9, 2017-12-21:
As far as I'm concerned, yes.

closed 2017-12-21 by JamesH65