raspberrypi/firmware #537: Official RPi Touchscreen rpi-f5406 swapxy

PapaSierraRomeo, 2016-02-04:
Hi,

I'm using the official RPi Touchscreen in portrait mode but am having problems getting the touchscreen to work in this orientation. My application, which is a Python/pygame app run from the command line,  does not need X Windows so the common means by which the touchscreen axes can be swapped (xinput set-prop 'FT5406 memory based driver' 'Evdev Axes Swap' 1 etc) won't help. I notice that touchscreen drivers like the ads7846 offer the ability to swapxy axes but the ft5406 driver has no similar parameters. 

Is there a way to achieve this with the rpi-ft5406 driver without using X-Windows? Would a swapxy parameter be a useful addition to the code?

Many thanks.

popcornmix, 2016-02-04:
I assume you can swap X/Y inside your game?
Presumably you are having to render your graphics to the screen with swapped X/Y and using swapped width/height from the framebuffer?

PapaSierraRomeo, 2016-02-04:
Thanks for responding. Swapping x and y in the game leaves a section of the screen which cannot be accessed owing to the 800x480 proportion so only about half of the screen is usable.

For rendering out the game I can use the display_rotate=1 command in /boot/config.txt which sorts out the display correctly and means that I don’t have to do anything special from my code.

The information contained in this e-mail is confidential and may
be subject to legal privilege. If you are not the intended recipient,
you must not use, copy, distribute or disclose the e-mail or any
part of its contents or take any action in reliance on it. If you have
received this e-mail in error, please e-mail the sender by replying
to this message. All reasonable precautions have been taken to
ensure no viruses are present in this e-mail and the sender cannot
accept responsibility for loss or damage arising from the use of this
e-mail or attachments.
.*****************************************************************************

 
**************************************************************
The information contained in this communication (including any attachments) is confidential and may be legally privileged. The views expressed may not be official policy, but the personal views of the originator. It is intended solely for the use of the individual or entity to whom it is addressed and others authorised to receive it. If you are not the intended recipient you are hereby notified that any disclosure, copying, distribution or taking action in reliance of the contents of this information is strictly prohibited and may be unlawful. If you have received it in error, please notify the sender by return e-mail and delete it from your system. The sender is neither liable for the proper, complete transmission of the information contained in this communication nor for any delay in its receipt.
 
Emails are not secure and cannot be guaranteed to be error free as they can be intercepted, amended, lost or destroyed, or contain viruses. Anyone who communicates with us by email is taken to accept these risks.
 
Please note we reserve the right to monitor and archive all e-mail communication through our internal and external networks so as to ensure compliance with internal policies and to protect our business.
 
This email is from De Beers UK Limited (Registered Number - 2054170), a private limited company forming part of the De Beers Group of Companies and incorporated in England and Wales with its Registered Office at 17 Charterhouse Street, London, EC1N 6RA. The Diamond Trading Company and the DTC are business names of De Beers UK Limited.

Ruffio, 2016-06-29:
@PapaSierraRomeo so your issue has been solved?

PapaSierraRomeo, 2016-06-30:
Not really. In the end I have gone the X-Windows route and am using the well established methods of rotating the touchscreen interface under X. Before that, and to try to maintain the command line only method of operation, I tried scaling and swapping the axes. This sort of worked but relied on the first touch of the screen when the program started being at 0,0 and only then would the correct co-ordinates be returned for subsequent position calls when other areas of the screen were touched. I tried to find a way to initialise the screen but couldn’t so ended up going the X-Windows route instead as forcing (or relying) on the user to initialise the display didn’t seem a viable solution.

 
 
 
This email is from De Beers UK Limited (Registered Number - 2054170), a private limited company forming part of the De Beers Group of Companies and incorporated in England and Wales with its Registered Office at 17 Charterhouse Street, London, EC1N 6RA. De Beers Auction Sales is a division of De Beers UK Limited

araczkowski, 2016-07-10:
I have very similar issue with touch on the official RPi Touchscreen and pygame, in my case the mode is landscape.
The touch works with some offset and no way to calibrate it..

JamesH65, 2017-12-21:
This should be fixed in the latest firmware/kernel where you can swap in X,Y etc via device tree.

closed 2017-12-21 by JamesH65