raspberrypi/firmware #538: Pi 2 freeze after some hours

tkalinows, 2016-02-05:
Hi, 
Like in topic - Pi2 freeze after smoe hours.  
- Both diode (red and green) is on (when freeze). RJ-45 diode blinking, but ssh not response. Nothing response.
- I'm not overclocking. 
- I turned off ipv6 (echo "install ipv6 /bin/true" >> /etc/modprobe.d/blacklist.conf; reboot) because it was last of below logs before freeze - but no result.  System steale freeze after few hours
- SD rather ok. No errors. Working quickly. It's new one.

kali@ka-iHome:~$ uname -a: Linux kali-iHome 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l GNU/Linux

last Logs:
kali@kali-iHome:~$ sudo tail /var/log/syslog -n 20
Feb  5 06:17:18 kali-iHome sendmail[650]: Starting Mail Transport Agent (MTA): sendmail.
Feb  5 06:17:18 kali-iHome systemd[1]: Started LSB: powerful, efficient, and scalable Mail Transport Agent.
Feb  5 06:17:18 kali-iHome apache2[649]: Starting web server: apache2AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this message
Feb  5 06:17:19 kali-iHome apache2[649]: .
Feb  5 06:17:19 kali-iHome systemd[1]: Started LSB: Apache2 web server.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Multi-User System.
Feb  5 06:17:19 kali-iHome systemd[1]: Reached target Multi-User System.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Graphical Interface.
Feb  5 06:17:19 kali-iHome systemd[1]: Reached target Graphical Interface.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Update UTMP about System Runlevel Changes...
Feb  5 06:17:19 kali-iHome systemd[1]: Started Update UTMP about System Runlevel Changes.
Feb  5 06:17:19 kali-iHome systemd[1]: Startup finished in 3.554s (kernel) + 20.095s (userspace) = 23.649s.
Feb  5 17:26:10 kali-iHome systemd[1]: Time has been changed
Feb  5 17:39:01 kali-iHome CRON[843]: (root) CMD (  [ -x /usr/lib/php5/sessionclean ] && /usr/lib/php5/sessionclean)
Feb  5 17:40:01 kali-iHome CRON[886]: (smmsp) CMD (test -x /etc/init.d/sendmail && /usr/share/sendmail/sendmail cron-msp)
Feb  5 17:41:00 kali-iHome systemd[1]: Starting Cleanup of Temporary Directories...
Feb  5 17:41:00 kali-iHome systemd[1]: Started Cleanup of Temporary Directories.
Feb  5 18:00:01 kali-iHome CRON[965]: (smmsp) CMD (test -x /etc/init.d/sendmail && /usr/share/sendmail/sendmail cron-msp)

kali@kali-iHome:~$ sudo tail /var/log/syslog -n 20
Feb  5 06:17:18 kali-iHome sendmail[650]: Starting Mail Transport Agent (MTA): sendmail.
Feb  5 06:17:18 kali-iHome systemd[1]: Started LSB: powerful, efficient, and scalable Mail Transport Agent.
Feb  5 06:17:18 kali-iHome apache2[649]: Starting web server: apache2AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this message
Feb  5 06:17:19 kali-iHome apache2[649]: .
Feb  5 06:17:19 kali-iHome systemd[1]: Started LSB: Apache2 web server.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Multi-User System.
Feb  5 06:17:19 kali-iHome systemd[1]: Reached target Multi-User System.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Graphical Interface.
Feb  5 06:17:19 kali-iHome systemd[1]: Reached target Graphical Interface.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Update UTMP about System Runlevel Changes...
Feb  5 06:17:19 kali-iHome systemd[1]: Started Update UTMP about System Runlevel Changes.
Feb  5 06:17:19 kali-iHome systemd[1]: Startup finished in 3.554s (kernel) + 20.095s (userspace) = 23.649s.
Feb  5 17:26:10 kali-iHome systemd[1]: Time has been changed
Feb  5 17:39:01 kali-iHome CRON[843]: (root) CMD (  [ -x /usr/lib/php5/sessionclean ] && /usr/lib/php5/sessionclean)
Feb  5 17:40:01 kali-iHome CRON[886]: (smmsp) CMD (test -x /etc/init.d/sendmail && /usr/share/sendmail/sendmail cron-msp)
Feb  5 17:41:00 kali-iHome systemd[1]: Starting Cleanup of Temporary Directories...
Feb  5 17:41:00 kali-iHome systemd[1]: Started Cleanup of Temporary Directories.
Feb  5 18:00:01 kali-iHome CRON[965]: (smmsp) CMD (test -x /etc/init.d/sendmail && /usr/share/sendmail/sendmail cron-msp)

any sugesttion? Can you help diagnose fault?
thank you in advance.
Tomasz Kalinowski.

popcornmix, 2016-02-05:
Afraid there is no useful information in the log.
Any USB devices or other peripherals connected?
Do you get crashes when using a clean raspbian install?
If not, then what do you install/change before it becomes unreliable?

tkalinows, 2016-02-05:
Thanks for answer,
No one to usb is not connected.
All was ok. I did not installed new apps. But i was updated, i.e: z-way-server.
So, i will try install new, clear raspbian :(
best regards
- Tomasz

Ruffio, 2016-05-20:
@tkalinows anything new?

tkalinows, 2016-05-20:
Hi!
All is clear. Hardware was wrong.
1. New Razzberry
2. New Rapberry

Now all is ok. Alle is working stable.
Best regards

closed 2016-05-20 by popcornmix