raspberrypi/firmware #539: omxil buffer flags and timestamp unreliable on audio output

fabled, 2016-02-05:
I'm trying to write an ALSA OMX IL component, that takes input from the vc audio_mixer / audio_decoder component's audio output. Basically it is to replace OMX.broadcom.audio_render component in the pipeline.

I've got it pretty much working already, but I'm trying to solve properly the AV sync and other issues. So far I've noted the following:
- the received buffers have always nFlags=0xe7 which is non sense
- the received nTimeStamp is not reliable either, it looks valid at times

Both fields are pretty essential to handle end-of-stream and a-v synchronization properly.

Additionally, the alsa component must be buffer supplier or things work very unreliably. Seems that the VC IL does not properly support being tunnel supplier for non-VC components.

6by9, 2016-02-07:
Have you defined OMX_SKIP64BIT as you'll find in all the sample IL apps (eg https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/hello_pi/Makefile.include#L2), and is defined for all the VC IL components?
Without it some fields in the buffer header struct get converted to 64bit and cause mismatches in offsets for some members of the struct.

fabled, 2016-02-08:
Ok. That was it. Thank you!

This means that OMX IL Headers are broken. They say that OMX_SKIP64BIT is only to support old compiler without 64-bit data types. But the 64-bit stuff is not aligned correctly causing the ABI to be different with or without the flag. Great.

closed 2016-02-08 by fabled

6by9, 2016-02-08:
Those are the headers as produced by Khronos. The review process really should have ensured the alignment was good, particularly on such a critical structure as OMX_BUFFERHEADERTYPE. Can't turn back time though.

I'm suggesting to others that we add either a #error or #warn if OMX_Core.h is included without that set. I can't see anyone sane really running a full IL system on Pi without interacting with the GPU, and the GPU expects that flag set.