raspberrypi/firmware #540: no debug kernel builds

l29ah, 2016-02-10:
The README specifies /debug and kernel_debug.img, but there aren't ones in the repo.
// Should i presume kernel{,7}.img are built with early printk?

popcornmix, 2016-02-10:
We used to ship a separate kernel with more kernel debug options (specifically what was needed to get function names in the backtrace of a panic) as the debug options reduced performance by about 10%.

However we have since got backtrace working without the performance hit and have dropped the separate kernel.

I've removed the reference from the README file.

popcornmix, 2016-02-10:
If you are interested these were the options that had a big performance hit:
https://github.com/raspberrypi/linux/commit/5412f282ae16b8872e0da31b576122d5b3440228

After they were removed the main kernel image could support the backtrace options.

closed 2016-02-20 by popcornmix