raspberrypi/firmware #541: Capturing video from camera with frame number annotations injects horizontal line

bennettrogers, 2016-02-11:
When capturing video from the camera module using [picamera](https://github.com/waveform80/picamera), a horizontal gray line appears in the image when annotate_frame_num is set to True. (The line isn't always visible at all resolutions, possibly due to aliasing).

I'm filing an issue here based on the [discussion](https://github.com/waveform80/picamera/issues/259) in the picamera repo. It sounds like it may not be intentional or necessary for the line to appear when only frame numbers are displayed.

popcornmix, 2016-02-12:
@6by9 - any thoughts?

6by9, 2016-02-12:
Yes, I'd commented on the Picamera issue saying it was a firmware issue and raise it here.
The line is always there assuming vc_image_fill does the right thing on YUV data. 
The logic adds it when adding frame number, shutter speed, analogue gain, or focus position. It's not needed for frame number so just needs a little rework.
I'll deal with it when I'm looking for a 5 minute thing to fix.

6by9, 2016-02-12:
Curious on resolution. If width is 1296, it only shows up if the vertical resolution is 794 or above. But width 1024 it shows up at >= 770 high. 
Not going to worry about it though as I have a patch that seems to do the right thing. I'll pass it across to be released

popcornmix, 2016-02-20:
@bennettrogers is this still an issue with latest rpi-update firmware?

bennettrogers, 2016-02-22:
This appears resolved in the latest firmware. Thank you!

closed 2016-02-22 by bennettrogers