raspberrypi/firmware #542: 64-bit core timer is being reset

9pi, 2016-02-14:
The Plan 9 OS on the pi2 uses the cortex-a7 generic timers to generate independent timer interrupts on each core.  All four generic timers depend on the shared 64-bit "core timer" at  address 4000001C which does the actual counting. After an embarrassingly long time chasing intermittent timing problems, I've discovered that the core timer is occasionally making a jump far into the future or even into the past.  This happens especially (but not only) during heavy SD card activity. Checking against the system timer at roughly 10ms intervals, with the core timer divider set to count at 1MHz, I observe changes like these:
`Before  ---- After`
`003D8B70D  20000017B`
`22E503B50   100000019`
`11820CAB7   100000140`
`132499D65   1000021EB`
The pattern seems to be that the core timer is being reset with the low word 0 and the high word 1 or 2.
One might suspect that a bad pointer somewhere in Plan 9 is causing an inadvertent write to the timer, but I've ruled this out by mapping that section of address space read-only.

Could the core timer be interfered with by the GPU firmware, or perhaps by something else on the SoC?  Can I do anything to prevent it?

P33M, 2016-02-15:
There's a bit of magic in the timer read registers. Are you sure you're respecting the read conditions detailed in https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2836/QA7_rev3.4.pdf (section 3.1)?

pelwell, 2016-02-15:
And in case you were in any doubt, only the ARM cluster can access these registers - not the VPU/GPU, nor the DMA or any other peripherals.

9pi, 2016-02-15:
If by "magic" you mean reading the low word first and then the high word, yes I'm doing that.  The read is always on core 0 with interrupts disabled, so I'm confident there are no intervening accesses.
If nothing else on the chip can touch these registers, could I be observing a hardware glitch?

pelwell, 2016-02-15:
> One might suspect that a bad pointer somewhere in Plan 9 is causing an inadvertent write to the timer, but I've ruled this out by mapping that section of address space read-only.

Are you sure it is only mapped once?

9pi, 2016-02-15:
> Are you sure it is only mapped once?

Good point.  It is only done intentionally in one place, but I will insert some code to check for accidental aliases.

9pi, 2016-03-01:
I can confirm that the address page 40000000 documented as containing the core timer registers is definitely only mapped once, and read-only.  However during my investigations I found a transient situation in Plan 9 kernel code where the translation table descriptor for VM address zero can contain rubbish.  It is guaranteed to be corrected before any user process is dispatched, and the kernel never dereferences page zero (doing so would cause a panic), so this ought to be benign.  But mysteriously (to me), if the rubbish value looks like a valid descriptor in certain ranges (eg 60000000 or E0000000) at an instant where USB controller registers (!) are being accessed, the core timer gets reset.

Having found the immediate cause of the reset, I can fix it by keeping bad values out of the page table. So I'm content for the issue to be closed.  But if anyone has a clue to how a bad mapping for a never-referenced page could cause this register reset, I'd love to hear it.

pelwell, 2016-03-01:
Thanks for the feedback. Closing.

closed 2016-03-01 by pelwell