raspberrypi/firmware #544: Illegal instruction after 61039e6976c206ea9805b4b2679125521346849d

kukabu, 2016-02-15:
Hello

I can't run vc utils after upgrade to 61039e6976c206ea9805b4b2679125521346849d on Raspberry Pi Model B Rev 2
``
$ vcgencmd
Illegal instruction
``

mirak011, 2016-02-15:
Same issue on Raspberry Pi Model A

pelwell, 2016-02-15:
We're working hard behind the scenes to get a Pi1-compatible build.

popcornmix, 2016-02-15:
It does seem the newer linaro toolchain won't build for armv6 with vfp.
It accepts the command line options but still produces an armv7 and vfpv3 instructions.
For now I've reverted the last firmware update, so rpi-update should be working again for Pi 1.

popcornmix, 2016-02-15:
A bit more investigation shows the newer toolchain can compile for armv6 happily, and produce suitable object files. However the linker uses armv7 versions of crt files, which causes the issue.

It looks like we need an older/custom toolchain to build for Pi 1.

XECDesign, 2016-02-15:
crosstool-ng 2d3c70d with this config might be a good starting point.
http://archlinuxarm.org/builder/xtools/xtools-dotconfig-v6

http://archlinuxarm.org/developers/distcc-cross-compiling

deborah-c, 2016-02-15:
I've been through something similar with compilation for M4 processors with the linaro chain; in general, the libraries don't seem to get switched to correct versions. (Calling into ARM-flavoured support routines on a processor that only understands Thumb really doesn't work well.)

I never took the time to get fully to the bottom of it, but switched to a prebuilt version that targeted the right architecture.

popcornmix, 2016-02-15:
Thanks XECDesign. I've built a toolchain based on that file (but with settings tweaked to match Jessie).
See: https://github.com/raspberrypi/tools/blob/master/configs/arm-rpi-4.9.3-linux-gnueabihf.config

There is also an rpi-update firmware built with it. Hopefully that is better that the last attempt...

mirak011, 2016-02-15:
Commit 96deffcf325c8c9ad47583d815a274bb7e955f35 works flawlessly, thank you!

kukabu, 2016-02-16:
I think now it solved. Thanks.

closed 2016-02-16 by kukabu