raspberrypi/firmware #545: RPi2 LiveTV Problem - VDR Restart itself over watchdog

vrabac, 2016-02-16:
Current situation: 
VDR 2.3.1 is running for one day without zapping to any channel, at the evening when I come home if I do zap to another channel, VDR will restart itself over watchdog (after 90 seconds). Sometimes when I arrive at home the picture will be in freeze state (but sometimes there is audio going without any glitches), and again after rezap with VDR will restart itself over watchdog.

It happens on FTA channels. 

I already contacted the VDR Developer and here is his statement:
If the RasPi is tuned to the same channel for an extended period of time, it shows exactly this behavior.
It can mostly be avoided by switching channels every couple of hours.
I don't think this is a VDR core problem, because it doesn't happen with other output devices (like the TT S2 6400, for instance).

Must be a firmware problem in the RasPi.

I am not including any other log then from VDR, as i am not sure what everything is needed to Debug this issue.

Here some Details:
Hardware: 
Raspberry PI 2 Model B 1GB
PSU 5V / 2A - Micro USB Stecker
SkyTV Ultimate V 2016 (DVB-S2 USB Card)
Transcend TS32GUSDHC10E Class 10 Extreme-Speed microSDHC 32GB 

SW:
ArchLinuxArm 
Kernel: 4.1.17-4-ARCH
linux-raspberrypi-4.1.17-4
raspberrypi-firmware-20160216-1
raspberrypi-firmware-bootloader-20160216-1
raspberrypi-firmware-bootloader-x-20160216-1
raspberrypi-firmware-examples-20160216-1
raspberrypi-firmware-tools-20160216-1

> Jan 24 23:49:07 vdrrpi vdr[234]: [234] switching to channel 6 (ServusTV HD Oesterreich)
Jan 25 16:04:28 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
Jan 25 16:04:34 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
Jan 25 16:04:39 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
Jan 25 16:04:45 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
Jan 25 16:04:51 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
Jan 25 16:04:57 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
Jan 25 16:05:02 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
Jan 25 16:43:13 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
..........
Jan 25 16:43:17 vdrrpi vdr[234]: [291] rpihddevice: OmxError(StreamCorrupt)
Jan 25 16:43:21 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
...........
Jan 25 20:40:00 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
Jan 25 20:40:06 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
Jan 25 20:40:11 vdrrpi vdr[234]: [1061] ERROR: TS packet not accepted in Transfer Mode
Jan 25 20:40:15 vdrrpi vdr[234]: [380] SVDRP < 127.0.0.1:56533 client connection accepted
Jan 25 20:40:15 vdrrpi vdr[234]: [380] switching to channel 6 (ServusTV HD Oesterreich)
Jan 25 20:42:15 vdrrpi vdr[234]: [234] PANIC: watchdog timer expired - exiting!
Jan 25 20:42:16 vdrrpi systemd[1]: vdr.service: Main process exited, code=exited, status=1/FAILURE
Jan 25 20:42:16 vdrrpi systemd[1]: vdr.service: Unit entered failed state.
Jan 25 20:42:16 vdrrpi systemd[1]: vdr.service: Failed with result 'exit-code'.
Jan 25 20:42:17 vdrrpi systemd[1]: vdr.service: Service hold-off time over, scheduling restart.
Jan 25 20:42:17 vdrrpi systemd[1]: Stopped Video Disk Recorder.
Jan 25 20:42:17 vdrrpi systemd[1]: Starting Video Disk Recorder..

popcornmix, 2016-02-16:
Pretty sure this is not a firmware issue. The firmware won't cause error messages to appear in an arm side log.

6by9, 2016-02-16:
It could be a firmware thing if it is decoding that throws an error and then the client doesn't behave nicely.
It looks like the decoder got data it wasn't happy with.
```Jan 25 16:43:17 vdrrpi vdr[234]: [291] rpihddevice: OmxError(StreamCorrupt)```

What is the timeline your log for when you interacted with it? Did you do something at 16:43:17, or is 20:40:15 when you "returned home" and changed channel? 
What codec is in use for your stream? MPEG2, MPEG4, or H264? At a guess you're using MPEG4 based on http://www.lyngsat.com/tvchannels/at/Servus-TV-Osterreich.html
You also seem to be switching to the same channel you were already on.

My thoughts are that a state transition might stall on the OMX decoder, and VDR just restarts as a way to get around it. Without significantly more logs from the OMX interactions it's not possible to say. A full dump of the thread states when the watchdog gives up would be even better if you felt like using GDB or similar.

vrabac, 2016-02-18:
I interacted with RPi2 on the evening at 20:45:15 and that was time I did returned home and changed channel. The stream from the ServusTV HD should be in the MPEG-4 format.

I'll try to create a dump of the thread states when the watchdog gives up, but from VDR itself there is no userdump crated because its killed by watchdog and if I disable VDR's watchdog nothing will happen. And I am not so gut to use VDR through the gdb fully.

Last night there was again problem that VDR restarts itself on zap. Though this time LiveTV was ok (video and audio in sync). But switch to other channels was not ok. Not that much to see in the log, as I was using --log=2 (which I did increased now to --log=3).

I did also compiled the VDR output plugin rpihddevice with "make DEBUG_BUFFERSTAT=1" to have more stuff logged.

Please not that here is the log from two days 16 and 17 Feb. I switched to another channel at Feb 17 20:13:47. Until that Audio/Video was ok on that ServusTV HD channel.

> $ sudo journalctl -f -u vdr
-- Logs begin at Thu 1970-01-01 01:00:06 CET. --
Feb 16 14:50:44 vdrrpi vdr[219]: [373] SVDRP listening on port 6419/tcp
Feb 16 14:50:47 vdrrpi vdr[219]: [219] retuning due to modification of channel 1 (ORF1 HD)
Feb 16 14:50:47 vdrrpi vdr[219]: [219] switching to channel 1 (ORF1 HD)
Feb 16 20:24:35 vdrrpi vdr[219]: [373] SVDRP < 127.0.0.1:48933 client connection accepted
Feb 16 20:24:35 vdrrpi vdr[219]: [373] switching to channel 6 (ServusTV HD Oesterreich)
Feb 16 20:24:35 vdrrpi vdr[219]: [373] SVDRP < 127.0.0.1:48933 connection closed
Feb 16 20:57:07 vdrrpi vdr[219]: [373] SVDRP < 127.0.0.1:48934 client connection accepted
Feb 16 20:57:07 vdrrpi vdr[219]: [373] SVDRP < 127.0.0.1:48934 connection closed
Feb 16 20:57:07 vdrrpi vdr[219]: [219] setting primary device to 3
Feb 16 20:57:07 vdrrpi vdr[219]: [219] switching to channel 6 (ServusTV HD Oesterreich)
Feb 17 20:13:47 vdrrpi vdr[219]: [373] SVDRP < 127.0.0.1:48935 client connection accepted
Feb 17 20:13:47 vdrrpi vdr[219]: [373] switching to channel 1 (ORF1 HD)
Feb 17 20:15:47 vdrrpi vdr[219]: [219] PANIC: watchdog timer expired - exiting!
Feb 17 20:16:08 vdrrpi systemd[1]: vdr.service: Main process exited, code=killed, status=11/SEGV
Feb 17 20:16:08 vdrrpi systemd[1]: vdr.service: Unit entered failed state.
Feb 17 20:16:08 vdrrpi systemd[1]: vdr.service: Failed with result 'signal'.
Feb 17 20:16:08 vdrrpi systemd[1]: vdr.service: Service hold-off time over, scheduling restart.
Feb 17 20:16:08 vdrrpi systemd[1]: Stopped Video Disk Recorder.
Feb 17 20:16:08 vdrrpi systemd[1]: Starting Video Disk Recorder...
Feb 17 20:16:08 vdrrpi vdr[678]: [678] VDR version 2.3.1 started
Feb 17 20:16:08 vdrrpi vdr[678]: [678] switched to user 'vdr'
Feb 17 20:16:08 vdrrpi vdr[678]: [678] codeset is 'ANSI_X3.4-1968' - unknown
Feb 17 20:16:08 vdrrpi vdr[678]: [678] override character table is 'ISO-8859-15'

vrabac, 2016-02-19:
Today same thing, today (as I installed debug packages) there is an coredump.
I am posting here relevant part here:

ilclient is an abstraction layer from OMX IL and was developed from  Broadcom ?? - maybe someone have an idea why the VDR is hanging there?

> #7  0x76efaf1c in do_futex_wait.constprop () from /usr/lib/libpthread.so.0
#8  0x76efb060 in __new_sem_wait_slow.constprop.0 () from /usr/lib/libpthread.so.0
#9  0x74d4eda4 in vcos_semaphore_wait (sem=0x74d64394 <vcos_thread_main+12>)
    at /home/dc4/projects/staging/userland/build/inc/interface/vcos/vcos_platform.h:254
#10 _vcos_thread_sem_wait () at /home/dc4/projects/staging/userland/build/inc/interface/vcos/vcos_platform.h:645
#11 vcos_generic_event_flags_get (flags=flags@entry=0x12cf2c8, bitmask=bitmask@entry=132, op=op@entry=5, suspend=suspend@entry=4294967295,
    retrieved_bits=retrieved_bits@entry=0x7eb7778c) at /home/dc4/projects/staging/userland/interface/vcos/generic/vcos_generic_event_flags.c:223
#12 0x75e5d14c in vcos_event_flags_get (retrieved_events=0x7eb7778c, suspend=4294967295, op=5, requested_events=132, flags=0x12cf2c8)
    at /opt/vc/include/interface/vcos/generic/vcos_generic_event_flags.h:118
#13 ilclient_wait_for_command_complete_dual (comp=comp@entry=0x12cf2b0, command=command@entry=OMX_CommandPortDisable, nData2=nData2@entry=130,
    other=other@entry=0x0) at ilclient.c:1250
#14 0x75e5d174 in ilclient_wait_for_command_complete (comp=comp@entry=0x12cf2b0, command=command@entry=OMX_CommandPortDisable,
    nData2=nData2@entry=130) at ilclient.c:1269
#15 0x75e5dec8 in ilclient_disable_port_buffers (comp=0x12cf2b0, portIndex=portIndex@entry=130, bufferList=<optimized out>,
    ilclient_free=ilclient_free@entry=0x0, private=private@entry=0x0) at ilclient.c:991
#16 0x75e45a34 in cOmx::StopVideo (this=0x12ad650) at omx.c:879

6by9, 2016-02-19:
Yes, ilclient was developed at Broadcom as a wrapper to simplify some of the IL calls.

VDR is executing a PortDisable on port 130 (video_decode input port), which also flushes the codec. Completing that command requires all buffers to be returned to the buffer supplier - by the looks of it, your client is likely to be the buffer supplier (it calls UseBuffer instead of AllocateBuffer).

You can try enabling a load of GPU logging by running ```vcgencmd set_logging level=0xc0``` beforehand, and ```sudo vcdbg log msg``` to grab the logs afterwards. That may give some more useful pointers. Please dump the full log on a github gist or similar, rather than cropping out the bits you think may be relevant.

Whilst not bullet-proof, ilclient and the IL services are pretty resilient as long as you communicate with them in the correct way. Running on other than the supported Raspbian distribution also means we're less likely to try to reproduce, but will give some guidance.

6by9, 2016-02-22:
I've not had the time to analyse the logs fully, but nothing stands out.

In all your examples you've left VDR tuned to "ServusTV HD Oesterreich". Does it happen on other channels? Preferably using a different codec, as this may be something in the MPEG4 side that is throwing a wobbly. I just want to reduce the number of variables.

Does VDR need a Pi2? If it will run under Raspbian I may try setting up a spare B+ with a DVB-S2 tuner on 28.2degE (as that is where my satellite dish points). It's likely to be slow and painful though if it really takes hours to fail.

reufer, 2016-02-22:
I've observed the error as well for a couple of times and it does not relate to the channel nor the video codec. But it always takes a day or more to happen, so it's hard to debug. Personally I use Gentoo for development also on the Raspberry, so it's definitely not about Arch-Linux.

VDR runs perfectly fine on a B+, I wrote down the basic steps for the installation: https://github.com/raspberrypi/firmware/issues/463#issuecomment-158791734

Thanks a lot for your help!

6by9, 2016-02-22:
OK, I'll try to set up a rig.

Is there a way to disable the watchdog that is resetting VDR on timeout? I want the system to freeze in the bad state so I can grab logs and examine state.

reufer, 2016-02-22:
Sure, just start vdr manually from the console and add `--watchdog=0` as an option. So, if you have a DVB device attached to your Pi and you want to control the OSD via telnet, the call could look like: `/usr/bin/vdr --watchdog=0 --cachedir=/var/cache/vdr --log=3.7 --video=/var/vdr/video/video --plugin='remote --port=tcp:3333 ' --plugin=rpihddevice`

Let me know if you have any further question. You'll find my email in rpihddevice's source, don't hesitate to PM me.

6by9, 2016-02-22:
Cheers. I'll have a look. Hopefully it's just a buffer gone astray somewhere which will block PortDisable completing.

vrabac, 2016-02-22:
Today there was no crash, but i was not able to switch from one another FTA mpeg4 to ServusTV HD channel. See this [gist](https://gist.github.com/anonymous/45c87b1d96ed3c94170c) paste,

PS: does reboot of RPi2 change the 'vcgencmd set_logging level=0xc0' to default ones? If not what would be default one 'level='

Thank you all for this time.

6by9, 2016-02-22:
Yes, a reboot resets the log level. Default is probably 0x40010000. 
It looks like you've done a reboot since you last set the logging level, as that isn't codec logging.

vrabac, 2016-02-22:
Ok, i set it back, there is something weird in 'dmesg -T'

> $ dmesg -T
$ [Mon Feb 22 23:43:40 2016] vc_mem_mmap: length 8192 is too big

I just stopped VDR, then flushed cache 'sudo /opt/vc/bin/vcgencmd cache_flush' , set the logging as written before 'sudo /opt/vc/bin/vcgencmd set_logging level=0xc0'
(level=192), started VDR and used 'sudo /opt/vc/bin/vcdbg log msg' to get the log. Nothing weird with VDR just to see if the log is correct now.

[Gist](https://gist.github.com/vrabac/6845d75a52ee58536995)

Let see tomorrow if there will be something.

vrabac, 2016-02-23:
So I disabled 'wachtdog=0' and tried today to switch channel, firstly I was 'ServusTV HD', then switched 'R9 Oesterreich HD'  and this was working, then switching back to 'ServusTV HD' and then it then I lost video/audio. As there is no watchdog any more active even after 5min nothing is happening. Here is VDR and 'vcdbg log msg'. Can you see anything?

[Gistlog](https://gist.github.com/vrabac/1648df45c451c13a2698)

6by9, 2016-02-23:
That's a much more useful log, but nothing looks out of place. I know I want everything, but the stack traces would have been really useful there to see what VDR was waiting for.

I'm just setting up a rig, though having realised I've got no spare satellite feeds in useful places it'll be DVB-T2 rather than DVB-S2. Seeing as this appears to be a decoder issue that shouldn't make any difference.

vrabac, 2016-02-24:
Can i do the stack traces over gdb if befor switching to another channel I hook gdb to VDR process? If you have an guidelines what to do I can try that.

6by9, 2016-02-24:
I'm fighting to get this to do anything useful.
It's installed and I now have all the Sandy Heath transmitter channels listed, but I can't get it to display anything on screen (HDMI nor 7" LCD screen), or get audio from radio channels. Any suggestions?
Omxplayer seems to be happy in playing files (MPEG2 or H264) so the GPU can do it.

(I can't quite believe that rpi-update has been omitted from the Raspbian Jessie Lite image either - that one's just annoying and seems an odd omission. I guess they want to use apt-get dist-upgrade instead).

6by9, 2016-02-24:
I'd followed your instructions which started VDR as a service, but seems not to have enabled the rpihddevice plugin. Trying to start it manually with -P rpihddevice I get a UI up, but it won't allow me to specify the telnet remote control.
I must confess to having taken top of tree instead of checking out your specified commit from last year, but would have thought ToT was the best thing to be testing.

6by9, 2016-02-24:
Some runes deduced thanks to Google.
So I'm starting it manually with 
```sudo vdr -P rpihddevice -P"remote -p tcp:3333"```
Is there a way of starting it as a service with those extra command line options?

I've got caught out that you're using vcgencmd to check for MPEG2 support - I'd hacked my firmware in the IL component rather than enable a key (whistles innocently!). "video format is not supported".
I now have BBC 1 up and running - I'm heading to bed!

(We'll see if you can convince me that VDR is better than tvheadend. I think not as I mainly want it as a recording box with multiple DVB-T, DVB-T2, and DVB-S2 tuners. It's on an x86 box at the moment as I don't think a Pi would cope).

reufer, 2016-02-25:
> Is there a way of starting it as a service with those extra command line options?

Sure, if you have used apt-get to install VDR, there should be init scripts installed already. But since I'm not a friend of  aptitude nor systemd, I'm afraid I can't help you here.

> I now have BBC 1 up and running - I'm heading to bed!

great to hear - hope you sleeped well!

> We'll see if you can convince me that VDR is better than tvheadend. I think not as I mainly want it as a recording box with multiple DVB-T, DVB-T2, and DVB-S2 tuners. It's on an x86 box at the moment as I don't think a Pi would cope

I'm not going to try to convince anybody. ;-) I'm running a separate VDR instance on my x86 server in the cellar for recording, my Pi's are solely used as clients - and therefore they're perfect. And VDR is just a great PVR software: Slim, fast, stable, without any glamour but - in my view - the highest usability I've ever seen for this kind of software.

vrabac, 2016-02-26:
So after 36 hours of no interaction, there was it again. Firstly switch to another channel was ok but LiveTV on that channel was not ok "ERROR: TS packet not accepted in Transfer Mode" then i switch back to ServusTV HD and the watchdog activated after 180 seconds.

There was no coredump created, I tried to interact with gdb but I am not sure if there is anything useful.

First 'vcdbg log msg' was when I have had  "ERROR: TS packet not accepted in Transfer Mode" the second one after watchdog.

After VDR restartet the ServusTV HD was working again just fine.

[Gist](https://gist.github.com/vrabac/0121248b46b88a62b62f)

vrabac, 2016-02-27:
Here so after 24h of no inetraction and trying to switch fort and back there is it again. Please look in gist inline comment: [gistpaste](https://gist.github.com/vrabac/b78a044d6bf0bcb4997c)

Do you need anymore log or is this enough what i posted?

6by9, 2016-02-27:
There's nothing obvious in any of these logs, so little point adding more of the same.
I have a rig set up, but not had a chance to actually run it to get more debug out. Give me some time.

vrabac, 2016-03-02:
On first switch watchdog was activated: [gistpaste](https://gist.github.com/vrabac/e54fd92acf162911f33e)

vrabac, 2016-03-11:
Today after arriving home and Powering TV i found the Live Image was still and the audio croping.
After I did try to switch to this same channel VDR crashed but not like before, even watchdog was not able to restart VDR.

Here is [gistlog](https://gist.github.com/vrabac/f7bb469184c6a80ce9ba): 
and here is [coredump](https://gist.github.com/vrabac/4d497e6ea01857dbe0ab): 

If you dont see anything, can i get somehow debug firmware to be able to create better log?

reufer, 2016-04-03:
While my productive Raspbperry 2 is running for weeks without any troubles, the new Raspberry 3 shows this issue quite frequently. I think it happens more often when stopping a replay, where buffer usage is hitting the maximum, whereas in live TV mode, the amount of used buffers is kept to a low value.

Let me know if you need some more debug logs!

rfollmann, 2016-04-03:
I obtain the same problem as reufer. Those crashes happen more often when using Raspberry Pi 3 rather than RPI 2.

reufer, 2016-05-09:
With kernel a1a3be140e1d7ebff64e0a81c2ec90056764621e I was not able to reproduce the issue anymore - I guess the modified locking did the trick!

popcornmix, 2016-05-10:
@vrabac is this still an issue for you with latest (rpi-update) firmware/kernel?

vrabac, 2016-05-16:
@popcornmix i'll try the new firmware/kernel ASAP and will write here the results, sorry for late replay!

vrabac, 2016-05-30:
Everything seems to be fine right now with the kernel [a1a3be1](https://github.com/raspberrypi/firmware/commit/a1a3be140e1d7ebff64e0a81c2ec90056764621e)

Thank you

closed 2016-05-30 by vrabac