raspberrypi/firmware #546: mpeg-2 playback crashes when pausing if used together with opengl

dividuum, 2016-02-18:
Hi.

I'm not exactly sure what happens, but together with a customer was able to reproduce the problem using stock raspbian (wheezy): Download this mpeg2 file: http://hubblesource.stsci.edu/sources/video/clips/details/images/hst_2.mpg

Then start in two different terminals:

    /opt/vc/src/hello_pi/hello_triangle/hello_triangle.bin

and

    omxplayer hst_2.mpg

Now pause and unpause the omxplayer video by pressing _space_ a few times. The video will get stuck and won't continue to play. You can terminate both programs but restarting won't work as they hang during initialization. I'm not entirely sure if the combination of playing mpeg2 and opengl is to blame here, but that's how we discovered this problem. Converting the same video to h264/mp4 solves the problem as well, so it's most likely mpeg2 related.

I'm using the following versions:

    Linux raspberrypi 4.1.17-v7+ #843 SMP Mon Feb 15 23:35:33 GMT 2016 armv7l GNU/Linux

    Feb 15 2016 23:38:07 
    Copyright (c) 2012 Broadcom
    version 263c9ef2f99450604d945968d75c013abf1fc841 (clean) (release)

    omxplayer - Commandline multimedia player for the Raspberry Pi
        Build date: Sat, 12 Sep 2015 00:49:19 +0000
        Version   : d99bd86 [master]
        Repository: https://github.com/popcornmix/omxplayer.git

on a Raspberry PI 2 with a Raspbian wheezy installed.

dividuum, 2016-03-17:
Can anyone confirm this problem? I tried it all my devices and always run into this. Is there a way to fix this?

popcornmix, 2016-03-17:
does ``omxplayer --nodeinterlace hst_2.mpg`` still have the problem?

dividuum, 2016-03-17:
That does indeed help and it seems to work. With deinterlacing it still stops. Any idea what causes the problem?

Test environment was:

    Raspbian Jessie on PI2

    Linux raspberrypi 4.1.19-v7+ #853 SMP Wed Mar 9 18:09:16 GMT 2016 armv7l GNU/Linux

    Mar  9 2016 18:11:14 
    Copyright (c) 2012 Broadcom
    version 3a754304b032a5298ee7889b179c667bbc75dec5 (clean) (release)

    omxplayer - Commandline multimedia player for the Raspberry Pi
        Build date: Sat, 02 Jan 2016 13:46:33 +0000
        Version   : f544084 [master]
        Repository: https://github.com/popcornmix/omxplayer.git

popcornmix, 2016-03-17:
Deinterlace uses the 3D hardware. In theory these should co-exist (and in fact must do as Kodi uses opengles for rendering GUI and supports deinterlace) but perhaps there is some conflict.

I'll try to reproduce.

popcornmix, 2016-03-17:
Can you try this test firmware:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_3deint.zip
Extract the contents to ``/boot`` directory (the one you see when sdcard is inserted into widnows).

dividuum, 2016-03-17:
Seems to work for my test case. I'll test it with a actual setup running at a customer site and reopen if I run into problems. Thanks! Is the change going to make it into the next official release?

closed 2016-03-17 by dividuum

popcornmix, 2016-03-18:
Yes, the changes should be in next firmware update.