raspberrypi/firmware #548: Latest commit breaks stereo passthrough vdr/rpihddevice

rfollmann, 2016-02-24:
I use raspberry pi HDMI output to TV and optical cable from TV to audio receiver. The latest commit (flac PCM) breaks stereo only TV channels in VDR using rpihddevice. Firmware before latest firmware is OK without an issues. With latest firmware I do not have sound on stereo channels, whereas dolby (2.0 and 5.1) channels are OK.

futant, 2016-03-03:
Hi, I'm not sure if this is the right forum or bug report but here goes. I'm using the stock raspbian and after the latest firmware updates I lost the DTS passthrough in Kodi. I verified this by downgrading the libraspberrypi* and raspberrypi-bootloader packages from 1.20160225-1 to 1.20160209-1 versions. After downgrade the DTS passthrough was once again working without any changes to Kodi configuration.
My setup: RP2b -> HDMI -> Sony KDL46HX TV -> Sony STR DB-790 AMP (dd + dts only, no dts-hd etc).

popcornmix, 2016-03-04:
@rfollmann Can you indicate exactly what software you are running on the Pi and what the audio codec is when playing stereo?

rfollmann, 2016-03-04:
Hello popcornmix,
Thank you for digging into this issue. I am using latest Raspbian version, kernel 4.1.18-v7+ #845. My setup is Raspberry Pi -> HDMI to LG 55UG8709 TV -> Optical to Panasonic SA HE-200 Audio receiver.
I am using VDR and rpihddevice set to audio passthrough. All Dolby channels are working (DD 2.0 or 5.1) perfectly. This seems to be AC3 codec. Simple stereo channels are not when using "flag compressed samples" firmware and firmware after. Audio codec should be 2ch MPEG (PCM?) in this case. Futant seems to face the same issue. With "flag compressed samples"-firmware (and above) I have most time no audio at all. Sometimes, it appears for just for a few seconds, before disappearing again for minutes. Another strange thing is: When playing a bit with pavucontrol, I can sometimes audio force to be present. 
My audio receiver has PCM Fix option. When switching this option on, audio is OK. However, in this case I do not have DD any more ;). So issue seems to be related to incoorect PCM audio packages.
As i said before, all firmwares before "flag compressed samples" one are perfectly OK. I hope, this may help you to find the root cause. Thank you for your help.
Kind regards,
Rudi

popcornmix, 2016-03-04:
When playing stereo (and it not working) can you ssh in and run
``vcgencmd hdmi_stream_channels``
I suspect vdr is incorrectly indicating passthrough when it wants PCM.

popcornmix, 2016-03-04:
You should see ``hdmi_stream_channels=1`` when passthrough is active and ``hdmi_stream_channels=0`` when passthrough is not active.

rfollmann, 2016-03-04:
You are right, popcornmix. When audio is not working (PCM): `hdmi_stream_channels=0`, when it is working I get `hdmi_stream_channels=1. Does this refer to VDR or to rpihddevice?

popcornmix, 2016-03-04:
I don't know the architecture of VDR/rphihddevice, but it looks like it is setting hdmi_stream_channels incorrectly. Best to report this to them.

rfollmann, 2016-03-05:
@popcornmix: If I execute `vcgencmd hdmi_stream_channels 1`, PCM audio does still not work. I thought it should?

futant, 2016-03-05:
Hi, I've been discussing about my issue also in  the raspbian [forum](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=109088&p=920690) but thought that this might be of interest. When using Kodi I get 'hdmi_stream_channels=1' even with the latest kernel + firmware and when the DTS passthrough is not working.
Rascas commented in the forums that the latest versions worked for him with a setup:
PI -> HDMI -> receiver -> HDMI -> TV (speakers connected to receiver).
My non-working setup is:
PI -> HDMI -> TV -> optical -> receiver (speakers connected to receiver).
So maybe this issue only affects S/PDIF?

The only commit that I could find that might have caused the issue is this: [github](https://github.com/raspberrypi/firmware/commit/565197e2f830388155dfd6ba713ea32a75697a26).

And again apologies if I am posting in the wrong bug report or thread, please advice if I should make a separate ticket about this. I don't mean to hijack this issue from rfollmann.

BR, futant

rfollmann, 2016-03-05:
@futant: This is precisly my setup.

rfollmann, 2016-03-07:
@popcornmix: I contacted both, Mr. VDR and Mr. RPIHDDEVICE. It seems to be rather a rpihddevice isseu than a VDR issue. Could you please provide kind of a switch (e.g. config.txt), which disables your firmware change unless we have found the root cause? I would need the newest firmware in order to make my bluetooth work. However, this one does not work for passthrough over SPDIF.

popcornmix, 2016-03-07:
> You are right, popcornmix. When audio is not working (PCM): hdmi_stream_channels=0, when it is working I get `hdmi_stream_channels=1. Does this refer to VDR or to rpihddevice?

This doesn't really make sense. The change made to firmware "spdif: flag compressed samples" is
```
-   if (datatype == CODEC_PCM)
+   if (datatype == CODEC_PCM && !C(hdmi_stream_channels))
       SPDIF_CLEAR( 1 );       // b=0: sample word contains linear PCM samples
    else
       SPDIF_SET( 1 );         // b=1: sample word contains compressed data
```

So there is no change in behaviour when hdmi_stream_channels=0. I can only assume that hdmi_stream_channels=1 was actually (incorrectly) true when the audio device was opened. Can you link to the code that sets hdmi_stream_channels in VDR/RPIHDDEVICE?

rfollmann, 2016-03-07:
@popcornmix: I contacted author of rpihddevice. I have already a quick and dirty workaround working. I am quite sure, the author will provide a fix soon. Sorry for all the inconvinience I may have caused and thank you for your help!

closed 2016-03-07 by rfollmann

futant, 2016-03-07:
@popcornmix Hi, should I open a separate ticket about my DTS passthrough with S/PDIF issue? It is not related to rpihdevice in any way. My case was also different from rfollmann's because the vcgencmd showed hdmi_stream_channels=1 also with the latest kernel package when the audio is broken. Is there any further info I can provide to help troubleshoot this issue?

popcornmix, 2016-03-07:
Create a new issue - I don't think yours was related to this.