raspberrypi/firmware #554: OpenELEC Latest Compile works on RPi2 but not RPI3

sghazagh, 2016-03-03:
I have successfully compiled the OpenELEC from source for RPi2 and the image has been created successfully for Raspberry Pi2/3.
However, the OpenELEC runs ok on RPi2 but for RPi3 it shows a rainbow colorful screen and stuck there.

I have updated the FAT partition with latest firmware from Raspberry Pi GIT to be sure the kernel and device tree of RPi3 is included but the result is same.

Any clue?

MilhouseVH, 2016-03-03:
This is an OpenELEC issue. The OpenELEC master branch doesn't yet include the RPi3 kernel commits. Consequently when you build the 4.4.3 kernel you're not creating `/flash/bcm2710-rpi-3-b.dtb`, and when you update your RPi3 with your new tar you end up using `/flash/bcm2710-rpi-3-b.dtb` from the previous installation, which is probably from 6.0.3 which is based on the 4.1.18 kernel. Result: rainbow screen.

For now, delete the 4.1.18 `/flash/bcm2710-rpi-3-b.dtb` and your RPi3 will boot with your new tar, but without WiFi or BT (it will think it is a RPi2).

To build OpenELEC master for RPi3 you need to include the three Feb 29 commits from https://github.com/raspberrypi/linux/commits/rpi-4.4.y. You will also need https://github.com/OpenELEC/OpenELEC.tv/pull/4808.

pelwell, 2016-03-03:
Thanks, @MilhouseVH.

@sghazagh Please close if you are happy with the answer.

sghazagh, 2016-03-03:
Thanks, @MilhouseVH

closed 2016-03-08 by sghazagh