raspberrypi/firmware #555: No commit for 4.1.18(-v7)+ #846

MrEngman, 2016-03-03:
Just done apt-get update/upgrade and kernel is now 4.1.18-v7+ #846 but there is no commit in raspberrypi/firmware or Hexxeh/rpi-firmware referencing build #846. 

Last two commits in raspberrypi/firmware, 35282eb 25 Feb and 9cd1c6c 29 Feb are for #845 and #848.

Where is the commit for #846?

Make compiling drivers for different commits rather awkward when they don't exist.

Ruffio, 2016-05-20:
@MrEngman is this still an issue? I can't figure out what you mean by #846. Do you have a direct link?

popcornmix, 2016-05-20:
Unfortunately I don't think we can help.
``#846`` was a custom build with Pi3 support allowing raspbian images to be produced before the Pi3 was announced. To avoid exposing information about Pi3 it wasn't pushed to the firmware repo.

However the firmware update on the Pi3 release date (``#848``) included additional fixes.
As the firmware repo is a single monotonically incrementing sequence of releases there is no way of pushing ``#846`` without causing more confusion.

closed 2016-05-20 by popcornmix