raspberrypi/firmware #556: vc4-kms-v3d overlay breaks OpenGL ES

WayneKeenan, 2016-03-03:
Hi,

I'm running into an issue when using OpenGLES that I can workaround by commenting out `dtoverlay=vc4-kms-v3d`  in `/boot/config.txt`

This is probably a known issue or can be put down to the work in progress nature of the OpenGL driver, but I thought I should log it just in case.

H/W:  Raspberry Pi 3 Model B
Linux `raspberrypi 4.1.18-v7+ #848 SMP Mon Feb 29 14:19:45 GMT 2016 armv7l GNU/Linux`
Dist: Rasbpian Jessie

Test command:
`/opt/vc/src/hello_pi/hello_teapot/hello_teapot.bin`

Result:
Running the above command displays no teapot and thus brews no tea.  
Instead, it displays the 'power-on RGB square' and in addition the process can't easily be killed.  
Other (SSH) sessions continue to function.

All the best
Wayne

WayneKeenan, 2016-03-03:
Should I have actually raised this against the raspberrypi/ linux repository instead?

popcornmix, 2016-03-03:
vc4-kms-v3d does not support GLES apps using dispmanx and  the firmware driver.
This is not expected to work.

WayneKeenan, 2016-03-04:
Ah ok, then please close this issue;  I had not appreciated the fact that OpenGLES wouldn't be okay even if the OpenGL accelerated desktop wasn't running.

I will raise a change request with the raspsi-config maintainers asking them to add a message on the 'Enable OpenGL desktop support" screen mentioning that enabling the option causes OpenGLES to no longer work either from the console or the desktop.

closed 2016-03-04 by WayneKeenan