raspberrypi/firmware #557: No secondary framebuffer available with VGA666

wypiki, 2016-03-03:
When enabling the Gert VGA-adapter with the following options in the config.txt file, I can play a video with omxplayer on the VGA-connected screen while using the desktop over HDMI, but there is no secondary framebuffer device to use with LXDE.

enable_dpi_lcd=1
display_default_lcd=0
dpi_group=2
dpi_mode=35
dtoverlay=vga666

popcornmix, 2016-03-04:
Only a single framebuffer is currently supported. You can use a secondary display for a dispmanx app, like omxplayer.

wypiki, 2016-03-18:
Ok. Are there any plans to support that? This would be important in my company because of the usage of two monitors over RDP. Or is there a way to do this over dispmanx?

alkemir, 2016-09-30:
Hi, if you mean to just clone the displays, this might help:

https://github.com/alkemir/gertCloner

wypiki, 2017-07-27:
Is there any progress on supporting a secondary display with a secondary framebuffer with the firmware driver or VC4? My company would buy a lot of Raspberry PIs if that would work, because it is ideal for working with remote desktops and X forwarding. Right now we have to use desktop PCs because of the seond screen.

JamesH65, 2018-06-29:
Multi framebuffer support is currently in testing. It will support HDMI, Composite, DSI and DPI. Not sure of timescales. 

Please see this forum thread to get hold of the beta releases. 

https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=216399