raspberrypi/firmware #558: raspian jessie doesn't boot since update with pwm_sdm: bugfixes

trulio, 2016-03-03:
raspbian-ua-netinst jessie doesn't boot since this update

pelwell, 2016-03-04:
@trulio Do you have any more information? Note that raspbian-ua-netinst is not a supported product, although we don't like to break things. You could create an issue [here](https://github.com/debian-pi/raspbian-ua-netinst/issues) as well.

@diederikdehaas Are you aware of this?

trulio, 2016-03-04:
Ok i think i will make a fresh install next week on my RPI2.
My raspian configuration was ok since 6 month with every rpi-update with successful update since mid 2015.
I'm using it in console mode as a mopidy server. 
i've read that there was some boot issues with RPI3 when booting without anything connected. 
May be this is a side effect due to integration of RPI3

popcornmix, 2016-03-04:
I have spotted something that could cause an issue with last firmware.
@trulio I've pushed an update - can you test?

trulio, 2016-03-04:
I will not be able to test it until monday or thuesday.
Moreover i don't know if i will be able to boot the system anymore.
I will make an image of my 2GB µsd card to be able to reproduce the issue and then I will test if overwriting boot files manually solves the problem.
Don't expect a feedback before thuesday.

popcornmix, 2016-03-04:
Manually copying start.elf/fixup.dat should work fine.

diederikdehaas, 2016-03-04:
@trulio With which version of the installer did you try it? And on which Pi version?
And about which update are you talking? The firmware package which gets installed by the net installer hasn't been updated in more then a year now.

trulio, 2016-03-04:
@diederikdehaas 
Hi , and thanks for your work.
You correctly guessed the situation.
This is a minimal jessie system build with one of the first rpi2 compliant install you delivered on rpi2.
The last successful rpi-update upgraded the kernel to 4.1.18 ( checked with uname -a after reboot). The default kernel launch 3.xx was of course disabled to allow boot on last kernel installed.

trulio, 2016-03-06:
Ok just tried to overwrite start.elf/fixup.dat of commit "arm_loader: Treat initial frequencies are invalid " and it still doesn't work. 
I will try tomorrow overwriting all boot files.

trulio, 2016-03-07:
ok now the system boot. thanks for all.
Trulio

pelwell, 2016-03-07:
OK to close?

trulio, 2016-03-07:
ok to close

closed 2016-03-07 by trulio