raspberrypi/firmware #559: VGA666 missing colour components on RPi3

tvjon, 2016-03-06:
I posted on the Q & A thread about RPI3's failure to work with the very useful vga666 adapter, but no response, so here's further information & perhaps someone can help?

Here are 2 runs of the output from the DPI interface on Pi3. The red component is absent, thus the display is either blue, or green, depending on SPI election in raspi-config, & thus unusable.

Iniitial testing at launch date with the available firmware updates produced no DPI activity at all, & omxplayer locked up until CTRL-c when asked to output to display 4.
 
The several subsequent updates have all produced the same error conditions described in the traces attached.

The final trace shows the same µSD card used on a RPI2 which properly displays BG & R components.

[image]

[image]

[image]

6by9, 2016-03-16:
I've just tried reproducing this, and looking at the GPIO state after enabling the vga666 overlay with raspi-gpio, I don't get the result I'm expecting:
```
pi@raspberrypi:~ $ sudo raspi-gpio get
BANK0 (GPIO 0 to 27):
  GPIO 00: level=1 fsel=0 alt=  func=INPUT
  GPIO 01: level=1 fsel=0 alt=  func=INPUT
  GPIO 02: level=0 fsel=6 alt=2 func=LCD_VSYNC
  GPIO 03: level=0 fsel=6 alt=2 func=LCD_HSYNC
  GPIO 04: level=0 fsel=6 alt=2 func=DPI_D0
  GPIO 05: level=0 fsel=6 alt=2 func=DPI_D1
  GPIO 06: level=0 fsel=6 alt=2 func=DPI_D2
  GPIO 07: level=0 fsel=6 alt=2 func=DPI_D3
  GPIO 08: level=0 fsel=6 alt=2 func=DPI_D4
  GPIO 09: level=0 fsel=6 alt=2 func=DPI_D5
  GPIO 10: level=0 fsel=6 alt=2 func=DPI_D6
  GPIO 11: level=0 fsel=6 alt=2 func=DPI_D7
  GPIO 12: level=0 fsel=6 alt=2 func=DPI_D8
  GPIO 13: level=0 fsel=6 alt=2 func=DPI_D9
  GPIO 14: level=1 fsel=2 alt=5 func=TXD1
  GPIO 15: level=1 fsel=2 alt=5 func=RXD1
  GPIO 16: level=0 fsel=0 alt=  func=INPUT
  GPIO 17: level=0 fsel=0 alt=  func=INPUT
  GPIO 18: level=0 fsel=0 alt=  func=INPUT
  GPIO 19: level=0 fsel=0 alt=  func=INPUT
  GPIO 20: level=0 fsel=0 alt=  func=INPUT
  GPIO 21: level=0 fsel=0 alt=  func=INPUT
  GPIO 22: level=0 fsel=0 alt=  func=INPUT
  GPIO 23: level=0 fsel=0 alt=  func=INPUT
  GPIO 24: level=0 fsel=0 alt=  func=INPUT
  GPIO 25: level=0 fsel=0 alt=  func=INPUT
  GPIO 26: level=0 fsel=0 alt=  func=INPUT
  GPIO 27: level=0 fsel=0 alt=  func=INPUT
```
GPIOs 2-21 should all be on alt=2, func=DPI or similar.
Same uSD card in a Pi2 does give the expected results.

That would imply that something else is reserving those GPIOs on a Pi3 somehow. I don't see anything obvious though.
My only thought is that Pi3 has uart1 enabled on GPIOs 14&15, so does the overlay abort once it sees those pins already assigned? Pi1 & Pi2 don't have GPIOs assigned to the UART driver, they seem to rely on the default dt-blob.bin the firmware runs to set the pin muxing appropriately.

@pelwell Guidance please? I haven't got the VGA666 attached, but that doesn't actually matter - it sets up the GPIOs and "sudo raspi-gpio get" shows you the appropriate state.
Can we simply add a line to disable uart1 from within vga666-overlay? I've failed to get the magic runes right for an overlay, but hacking it out of bcm2710-rpi-3-b.dtb means that I get the pin config I'm expecting for VGA666, so I think my diagnosis is right.

pelwell, 2016-03-16:
@6by9 Your diagnosis sounds correct. Although the pins 2-13 have been left on alt2, they will be unclaimed.

I'm reluctant to add automatic uart1 disabling to the vga666 overlay because of the varied UART usage on the Pi models. I will be changing the firmware sometime this week to automatically configure pins 14 & 15 for uart1 on the Pi3, at which point I could take the default pin assignment for it out of the base DTBs. But until then, and to match uart0, I've added a dtparam to enable or disable uart1.  You would use it by adding `dtparam=uart1=off` to your config.txt. It will be in the next update, or you can built it yourself.

6by9, 2016-03-16:
Thanks @pelwell - that sounds like a plan. I'll retest tonight with ```dtparam=uart1=off``` to confirm. I think I messed up my system last night - I'd updated to 4.4 so had a mixture of old and new stuff.

I'll also see if I can get a DPI display working via overlay. I previously had it setting all the pin muxing correctly, but didn't get any output up on the screen. The only thing I could see left was drive current which had been set to 8mA in the dt-blob.bin. I couldn't see any way to set that from overlays/dt.

There was a post on the forum last night that confirmed that adding ```dtoverlay=pi3-disable-bt-overlay``` also brought vga666 back to life, which is doing a similar thing. That would indicate it was just my system on a messed up 4.4 that was causing problems. (https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=94424&start=75#p930066)

tvjon, 2016-03-16:
6by9, just in case it helps, I've just done:

sudo BRANCH=next rpi-update

to 4.4 & without doing any changes from the setup of last night, VGA666 is still working.

pelwell, 2016-03-16:
The bcm2835-gpio (combined gpio/pinmux) driver doesn't expose bindings for setting drive strength. 
I was slightly surprised that `pi3-disable-bt` didn't suffer the same problem since with that overlay uart0 grabs 14&15, but it does. Although the uart1 node is probed before the leds (which we are (ab)using to allocate the pins for use), the uart0 node is probed afterwards. This allows all of the pins to be claimed for vga666, and denies uart0 its pins.

Another wrinkle is that a pinmux failure doesn't prevent a node from being created - it just doesn't get its pins configured. This means that although 14 & 15 are assigned to the DPI function, changing that to alt0 at runtime allows uart0 to work again - it waits with its pins unconnected.

6by9, 2016-03-16:
Such fun!
I guess there has to be an ordering of some form, and it won't always put things into the order you ideally want for all situations - there isn't such a thing.

The pinmux failure seems to be an odd one though in that it seems to abort mux setup on the first failure. In that case does it release the pin assignments already done (though not undoing the muxing), but still create the node, or do those remain assigned?

I sort of expected that bcm2835-gpio wouldn't expose drive strength. It appears to be a global per GPIO bank, so Linux isn't going to understand that concept. 
I'll persevere with my DPI overlay. Perhaps adding a config.txt option for setting the minimum drive strength of each bank would be a suitable solution of removing the dependence on dt-blob.bin, but that is presuming that drive strength is the issue - a few more tests before I can say that for definite.

Thanks for the advice.

pelwell, 2016-03-16:
With uart1 still enabled and with the vga666 overlay loaded (unsuccessfully):
```
pi@raspberrypi:~$ sudo cat /sys/kernel/debug/pinctrl/3f200000.gpio/pins
registered pins: 54
pin 0 (gpio0) function gpio_in in hi; irq 166 (none)
pin 1 (gpio1) function gpio_in in hi; irq 167 (none)
pin 2 (gpio2) function alt2 in lo; irq 168 (none)
pin 3 (gpio3) function alt2 in lo; irq 169 (none)
pin 4 (gpio4) function alt2 in lo; irq 170 (none)
pin 5 (gpio5) function alt2 in lo; irq 171 (none)
pin 6 (gpio6) function alt2 in lo; irq 172 (none)
pin 7 (gpio7) function alt2 in lo; irq 173 (none)
pin 8 (gpio8) function alt2 in lo; irq 174 (none)
pin 9 (gpio9) function alt2 in lo; irq 175 (none)
pin 10 (gpio10) function alt2 in lo; irq 176 (none)
pin 11 (gpio11) function alt2 in lo; irq 177 (none)
pin 12 (gpio12) function alt2 in lo; irq 178 (none)
pin 13 (gpio13) function alt2 in lo; irq 179 (none)
pin 14 (gpio14) function alt5 in hi; irq 180 (none)
pin 15 (gpio15) function alt5 in hi; irq 181 (none)
pin 16 (gpio16) function gpio_in in lo; irq 182 (none)
pin 17 (gpio17) function gpio_in in lo; irq 183 (none)
pin 18 (gpio18) function gpio_in in lo; irq 184 (none)
pin 19 (gpio19) function gpio_in in lo; irq 185 (none)
pin 20 (gpio20) function gpio_in in lo; irq 186 (none)
pin 21 (gpio21) function gpio_in in lo; irq 187 (none)
pin 22 (gpio22) function gpio_in in lo; irq 188 (none)
pin 23 (gpio23) function gpio_in in lo; irq 189 (none)
pin 24 (gpio24) function gpio_in in lo; irq 190 (none)
pin 25 (gpio25) function gpio_in in lo; irq 191 (none)
pin 26 (gpio26) function gpio_in in lo; irq 192 (none)
pin 27 (gpio27) function gpio_in in lo; irq 193 (none)
pin 28 (gpio28) function gpio_in in lo; irq 194 (none)
pin 29 (gpio29) function gpio_in in hi; irq 195 (none)
pin 30 (gpio30) function gpio_in in lo; irq 196 (none)
pin 31 (gpio31) function gpio_in in lo; irq 197 (none)
pin 32 (gpio32) function alt3 in hi; irq 198 (none)
pin 33 (gpio33) function alt3 in hi; irq 199 (none)
pin 34 (gpio34) function alt3 in lo; irq 200 (none)
pin 35 (gpio35) function alt3 in hi; irq 201 (none)
pin 36 (gpio36) function alt3 in hi; irq 202 (none)
pin 37 (gpio37) function alt3 in hi; irq 203 (none)
pin 38 (gpio38) function alt3 in hi; irq 204 (none)
pin 39 (gpio39) function alt3 in hi; irq 205 (none)
pin 40 (gpio40) function alt0 in lo; irq 206 (none)
pin 41 (gpio41) function alt0 in lo; irq 207 (none)
pin 42 (gpio42) function alt0 in hi; irq 208 (none)
pin 43 (gpio43) function alt0 in lo; irq 209 (none)
pin 44 (gpio44) function gpio_in in hi; irq 210 (none)
pin 45 (gpio45) function gpio_in in hi; irq 211 (none)
pin 46 (gpio46) function gpio_in in hi; irq 212 (none)
pin 47 (gpio47) function gpio_in in hi; irq 213 (none)
pin 48 (gpio48) function alt0 in lo; irq 214 (none)
pin 49 (gpio49) function alt0 in hi; irq 215 (none)
pin 50 (gpio50) function alt0 in hi; irq 216 (none)
pin 51 (gpio51) function alt0 in hi; irq 217 (none)
pin 52 (gpio52) function alt0 in hi; irq 218 (none)
pin 53 (gpio53) function alt0 in hi; irq 219 (none)
```
```
pi@raspberrypi:~$ sudo cat /sys/kernel/debug/pinctrl/3f200000.gpio/pinmux-pins
Pinmux settings per pin
Format: pin (name): mux_owner gpio_owner hog?
pin 0 (gpio0): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 1 (gpio1): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 2 (gpio2): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 3 (gpio3): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 4 (gpio4): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 5 (gpio5): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 6 (gpio6): (MUX UNCLAIMED) (GPIO UNCLAIMED)
[... 28 lines cut ...]
pin 35 (gpio35): 3f300000.mmc (GPIO UNCLAIMED) function alt3 group gpio35
pin 36 (gpio36): 3f300000.mmc (GPIO UNCLAIMED) function alt3 group gpio36
pin 37 (gpio37): 3f300000.mmc (GPIO UNCLAIMED) function alt3 group gpio37
pin 38 (gpio38): 3f300000.mmc (GPIO UNCLAIMED) function alt3 group gpio38
pin 39 (gpio39): 3f300000.mmc (GPIO UNCLAIMED) function alt3 group gpio39
pin 40 (gpio40): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 41 (gpio41): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 42 (gpio42): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 43 (gpio43): 3f201000.uart (GPIO UNCLAIMED) function alt0 group gpio43
pin 44 (gpio44): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 45 (gpio45): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 46 (gpio46): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 47 (gpio47): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 48 (gpio48): 3f202000.sdhost (GPIO UNCLAIMED) function alt0 group gpio48
pin 49 (gpio49): 3f202000.sdhost (GPIO UNCLAIMED) function alt0 group gpio49
pin 50 (gpio50): 3f202000.sdhost (GPIO UNCLAIMED) function alt0 group gpio50
pin 51 (gpio51): 3f202000.sdhost (GPIO UNCLAIMED) function alt0 group gpio51
pin 52 (gpio52): 3f202000.sdhost (GPIO UNCLAIMED) function alt0 group gpio52
pin 53 (gpio53): 3f202000.sdhost (GPIO UNCLAIMED) function alt0 group gpio53
```
This shows that although pins 2-13 have been left in alt2, they are free. I would prefer that they were returned to the input state - I'll add that to the list of things to look at.

In reading the pinmux sysfs output it may help to know that for horrible reasons (*) the GPIO and PINMUX worlds are independent - claiming a pin in one has no effect on the other, hence the two columns.

(*) It was convenient for NVidia to assign a group of pins a function without preventing some of them for being used for GPIO.

tvjon, 2016-03-16:
Here's my results for comparison, along with raspi-gpio-get at the end:

pi@j1 ~ $ sudo cat /sys/kernel/debug/pinctrl/3f200000.gpio/pins
registered pins: 54
pin 0 (gpio0) function gpio_in in hi; irq 166 (none)
pin 1 (gpio1) function gpio_in in hi; irq 167 (none)
pin 2 (gpio2) function alt2 in lo; irq 168 (none)
pin 3 (gpio3) function alt2 in lo; irq 169 (none)
pin 4 (gpio4) function alt2 in lo; irq 170 (none)
pin 5 (gpio5) function alt2 in lo; irq 171 (none)
pin 6 (gpio6) function alt2 in lo; irq 172 (none)
pin 7 (gpio7) function alt2 in lo; irq 173 (none)
pin 8 (gpio8) function alt2 in lo; irq 174 (none)
pin 9 (gpio9) function alt2 in lo; irq 175 (none)
pin 10 (gpio10) function alt2 in lo; irq 176 (none)
pin 11 (gpio11) function alt2 in lo; irq 177 (none)
pin 12 (gpio12) function alt2 in lo; irq 178 (none)
pin 13 (gpio13) function alt2 in lo; irq 179 (none)
pin 14 (gpio14) function alt2 in lo; irq 180 (none)
pin 15 (gpio15) function alt2 in lo; irq 181 (none)
pin 16 (gpio16) function alt2 in lo; irq 182 (none)
pin 17 (gpio17) function alt2 in lo; irq 183 (none)
pin 18 (gpio18) function alt2 in lo; irq 184 (none)
pin 19 (gpio19) function alt2 in lo; irq 185 (none)
pin 20 (gpio20) function alt2 in lo; irq 186 (none)
pin 21 (gpio21) function alt2 in lo; irq 187 (none)
pin 22 (gpio22) function gpio_in in lo; irq 188 (none)
pin 23 (gpio23) function gpio_in in lo; irq 189 (none)
pin 24 (gpio24) function gpio_in in lo; irq 190 (none)
pin 25 (gpio25) function gpio_in in lo; irq 191 (none)
pin 26 (gpio26) function gpio_in in lo; irq 192 (none)
pin 27 (gpio27) function gpio_in in lo; irq 193 (none)
pin 28 (gpio28) function gpio_in in lo; irq 194 (none)
pin 29 (gpio29) function gpio_in in hi; irq 195 (none)
pin 30 (gpio30) function gpio_in in lo; irq 196 (none)
pin 31 (gpio31) function gpio_in in lo; irq 197 (none)
pin 32 (gpio32) function gpio_in in hi; irq 198 (none)
pin 33 (gpio33) function gpio_in in hi; irq 199 (none)
pin 34 (gpio34) function alt3 in hi; irq 200 (none)
pin 35 (gpio35) function alt3 in hi; irq 201 (none)
pin 36 (gpio36) function alt3 in hi; irq 202 (none)
pin 37 (gpio37) function alt3 in hi; irq 203 (none)
pin 38 (gpio38) function alt3 in hi; irq 204 (none)
pin 39 (gpio39) function alt3 in hi; irq 205 (none)
pin 40 (gpio40) function alt0 in hi; irq 206 (none)
pin 41 (gpio41) function alt0 in hi; irq 207 (none)
pin 42 (gpio42) function alt0 in lo; irq 208 (none)
pin 43 (gpio43) function alt0 in hi; irq 209 (none)
pin 44 (gpio44) function gpio_in in hi; irq 210 (none)
pin 45 (gpio45) function gpio_in in hi; irq 211 (none)
pin 46 (gpio46) function gpio_in in hi; irq 212 (none)
pin 47 (gpio47) function gpio_in in hi; irq 213 (none)
pin 48 (gpio48) function alt0 in lo; irq 214 (none)
pin 49 (gpio49) function alt0 in hi; irq 215 (none)
pin 50 (gpio50) function alt0 in hi; irq 216 (none)
pin 51 (gpio51) function alt0 in hi; irq 217 (none)
pin 52 (gpio52) function alt0 in hi; irq 218 (none)
pin 53 (gpio53) function alt0 in hi; irq 219 (none)

pi@j1 ~ $ sudo cat /sys/kernel/debug/pinctrl/3f200000.gpio/pinmux-pins
Pinmux settings per pin
Format: pin (name): mux_owner gpio_owner hog?
pin 0 (gpio0): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 1 (gpio1): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 2 (gpio2): soc:leds (GPIO UNCLAIMED) function alt2 group gpio2
pin 3 (gpio3): soc:leds (GPIO UNCLAIMED) function alt2 group gpio3
pin 4 (gpio4): soc:leds (GPIO UNCLAIMED) function alt2 group gpio4
pin 5 (gpio5): soc:leds (GPIO UNCLAIMED) function alt2 group gpio5
pin 6 (gpio6): soc:leds (GPIO UNCLAIMED) function alt2 group gpio6
pin 7 (gpio7): soc:leds (GPIO UNCLAIMED) function alt2 group gpio7
[... 81 lines cut ...]
  GPIO 31: level=0 fsel=0 alt=  func=INPUT
  GPIO 32: level=1 fsel=0 alt=  func=INPUT
  GPIO 33: level=1 fsel=0 alt=  func=INPUT
  GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK
  GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
  GPIO 40: level=1 fsel=4 alt=0 func=PWM0
  GPIO 41: level=1 fsel=4 alt=0 func=PWM1
  GPIO 42: level=1 fsel=4 alt=0 func=GPCLK1
  GPIO 43: level=1 fsel=4 alt=0 func=GPCLK2
  GPIO 44: level=1 fsel=0 alt=  func=INPUT
  GPIO 45: level=1 fsel=0 alt=  func=INPUT
BANK2 (GPIO 46 to 53):
  GPIO 46: level=1 fsel=0 alt=  func=INPUT
  GPIO 47: level=1 fsel=0 alt=  func=INPUT
  GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
  GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
  GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
  GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
  GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
  GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3
pi@j1 ~ $

6by9, 2016-03-16:
Yup, all looks good with dtparam=uart1=off for VGA666. I guess a PR is required to update the docs and state that dtparam=uart1=off is required on Pi3.

(Still can't get my DPI display running via an overlay. Set the pin config in dt-blob.bin and it works. Set the same config from the overlay, raspi-gpio confirms the alt2 settings no pull, but no image on the screen. Then again, I've failed to get DPI display running at all on a Pi3 so far)

Ruffio, 2017-01-07:
@tvjon Can this issue be closed?

jonasbits, 2017-08-05:
working on rpi3 with lines:

dtparam=uart1=off 
dtoverlay=vga666
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=82

closed 2017-08-05 by tvjon