raspberrypi/firmware #561: Latest kernel/firmware breaks DTS passthrough with S/PDIF in Kodi

futant, 2016-03-07:
After updating the libraspberrypi0, libraspberry-bin and raspberrypi-bootloader packages from 1.20160209-1 to 1.20160225-1 the DTS passthrough stops working. Using the latest vanilla Raspbian image + installed Kodi using aptitude. Downgrading the packages to previous version enables DTS passthrough again without any changes in Kodi configuration. When playing DTS or DTS-HD files with the latest version (4.1.18-v7+ #846) there is no sound from the receiver and DTS is not recognized. With previous version (4.1.17-v7+ #838) the receiver indicates DTS3/2.1 stream.

My setup is RP2B -> HDMI -> Sony KDL 46HX755 tv -> optical -> Sony STR DB-790 receiver (dts and dd only, no hd support). MMAL on, OMX off in Kodi.

The issue was also discussed in [Raspbian forums](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=109088&start=425) where the user Rascas commented that the latest kernel/firmware was working for him with the setup: PI -> HDMI -> TV -> optical -> receiver (speakers connected to receiver), so it seems to affect only the cases where optical link is in place.

The command `vcgencmd hdmi_stream_channels` returns `hdmi_stream_channels=1` with both versions when DTS stream is playing. I have tested with multiple videos containing DTS or DTS-HD streams.

This issue currently prevents me from having up-to-date packages in Raspbian, I've had to hold the packages in aptitude for now.

There is more previous discussion related to this issue in here [Github](https://github.com/raspberrypi/firmware/issues/548).

popcornmix, 2016-03-07:
If you enable omxplayer, what is the behaviour with DTS passthrough before and after the firmware change?
With omxplayer disabled, what is the behaviour with AC3 passthrough before and after the firmware change?

futant, 2016-03-07:
omxplayer enabled, mmal disabled - no DTS passthrough in older or new firmware.
omxplayer enabled, mmal enabled - DTS passthrough working in older firmware, no DTS passthrough in new firmware and the video was showing it was actually using mmal.
omxplayer disabled, mmal enabled - AC3 passthrough working in older and new firmware.

popcornmix, 2016-03-07:
I think the problem is that your TV doesn't support (probably deliberately) passing DTS audio out through optical.
The manual explicitly mentions Dolby (AC3) but not DTS:
http://pdf.crse.com/manuals/AE7L100511/ENG/c_extsnd_digaout.html
and support for DTS is very rare:
http://www.cnet.com/uk/news/20-tvs-tested-which-sets-can-pass-surround-sound-to-a-sound-bar/

The recent firmware update was a correct fix, to signal compressed audio when we are outputting compressed audio, and it has fixed passthrough for numerous users.
However your TV is deliberately disabling audio when DTS is now detected (presumably for licensing reasons).

futant, 2016-03-07:
Ok, your explanation makes sense and I've seen a lot of similar comments about DTS support being rare in TVs. Still... would it be possible to enable the previous functionality of your patch via some configuration switch (like hdmi_force_edid_audio=1 which by the doesn't help in this situation)?
After all my current hardware is perfectly capable of DTS passthrough, it's just the software change that is preventing this undocumented feature from working. It would feel really stupid to buy a new HDMI receiver just to be able to update Raspberry Pi firmwares in the future.

popcornmix, 2016-03-07:
Let me see if there's any convenient config variable that could be used...

futant, 2016-03-07:
Thank you very much! I'm out of town for the next 2-3 days and won't be able to comment on this thread or test any changes before I get back.

popcornmix, 2016-03-09:
Latest rpi-update allows the compressed audio flag to be overridden.
``hdmi_audio_config=0x200`` should bring back the old behaviour (and also work with omxplayer).

futant, 2016-03-09:
I updated the kernel and fw with rpi-update and tested this with build #853 [Hexxeh / rpi firmware 2a7eb4f](https://github.com/Hexxeh/rpi-firmware/commit/2a7eb4fc4cef07906c36e9adcf76f053daabe371). DTS passthrough is not working without the config param `hdmi_audio_config=0x200`. After adding the param DTS passthrough works with both MMAL and OMXPLAYER (that's a nice bonus!). Also AC3 and couple of other formats continued to work without any regressions. Looks good to me! Closing the issue, thank you for really good support.

closed 2016-03-09 by futant