raspberrypi/firmware #562: Broken #include statements in /opt/vc/include

fordfrog, 2016-03-08:
while trying to compile kodi on rpi 2, i've encountered many errors related to #include files not being found. looking at the source it seems some include files were move around but the paths were not adjusted. these are the correct paths for current code:

#include "interface/vcos/pthreads/vcos_platform_types.h"
#include "interface/vcos/pthreads/vcos_platform.h"
#include "interface/vmcs_host/linux/vchost_config.h"

Ruffio, 2016-08-08:
@fordfrog how about submitting a PR for this?

popcornmix, 2016-08-08:
Not sure this is still an issue after https://github.com/raspberrypi/userland/pull/298

Ruffio, 2016-08-08:
@fordfrog can you please upgrade and confirm if your issue has been resolved? Thanks.

fordfrog, 2016-09-19:
@Ruffio i do not have rpi with gentoo anymore so unfortunately i cannot test it.

Ruffio, 2016-09-19:
@fordfrog since you you are the only one with this issue for 6 months, I suggest that you close this issue. Thanks.

closed 2016-09-19 by popcornmix