raspberrypi/firmware #564: rpi-display.dts - swapxy for touchscreen

bungee-, 2016-03-09:
I installed new raspbian jessie-lite and some common hardware is now already supported "out of the box" which is nice. 
So when using rpi-display overlay it adds the ads7864 touchscreen (hardware is there) support. Problem is because when using the display with 90 or 270 rotation that is not translated to touchscreen input as well so the xy axes are swapped.

Can the swapxy from the ads7864 overlay be added please.

pelwell, 2016-03-09:
It should be trivial - I'll get it done tomorrow.

pelwell, 2016-03-09:
OK, I lied - it's done. You just missed an rpi-update though, so it may take a few days to filter out.

bungee-, 2016-03-09:
I just did rpi-update and version now is:
4.1.19-v7+ #853 SMP Wed Mar 9 18:09:16 GMT 2016 armv7l GNU/Linux

001331.455: Loaded overlay 'rpi-display'
001331.471: dtparam: rotate=90
001332.180: dtparam: speed=32000000
001332.881: dtparam: debug=1
001333.628: dtparam: swapxy=1
**001335.710: Unknown dtparam 'swapxy' - ignored**
001447.486: dtparam: arm_freq=900000000
001477.602: dtparam: core_freq=250000000
001485.063: dtparam: cache_line_size=64

Still no support for xy swap in dts, sadly.

bungee-, 2016-03-09:
No problem. As long as the option will be there. :D cheers.

closed 2016-06-06 by pelwell