raspberrypi/firmware #565: wrong header location for pthreads-headers (while gst-omx)

Dambe, 2016-03-11:
I try building GStreamer from source and get stuck because of build errors due to a missing vcos_platform_types.h / vcos_platform.h when building gst-omx.

I found a similar issue here:
https://github.com/raspberrypi/firmware/issues/34

I got the Raspbian image from 2016-02-26 and it seems that this issue is still not fixed.

When looking in the /opt/vc/include/interface/vcos/vcos_*.h files the include paths are still not correct.
They should be "./pthreads/..." I guess?

Copying the files in /opt/vc/include/interface/vcos/pthreads to /opt/vc/include/interface/vcos seems like a dirty hack to me ...

ED6E0F17, 2016-03-11:
+1 for fixing this. Is it still necessary to track a private upstream repo?

popcornmix, 2016-03-11:
It's not just that. vcos can be built for other platforms and the mechanism is to use the include path to describe the platform it's being built for. While most users are using the linux/pthreads paths, I am aware of users who build using the win32 and bionic paths.
We also build the same code on the GPU using the threadx paths.
So just hard coding in paths would break building for other use cases.

Now we may be able to decide that the linux/pthreads case is common enough that some ifdeffery that defaults to the paths you want is possible, as long as it doesn't break building for other platforms.

However the current system of setting two include directories in your make system doesn't seem unreasonable.

closed 2018-06-29 by JamesH65